genodelabs/genode #1667: nic: support Zynq-7000

m-stein, 2015-09-03:


m-stein, 2015-09-03:
@ValiValpas It would be nice to have a known-to-work test for Zynq NIC first. As the commit adapts ``network_test_nic.inc``, I tried ``network_test_nic_raw``, ``network_test_nic_raw_client``, and ``network_test_nic_raw_bridge_client`` but they all fail to compile.

For further development it would be nice if you could use my branch [1667_nic_support_Zynq_7000](https://github.com/m-stein/genode/tree/1667_nic_support_Zynq_7000) as base.

ValiValpas, 2015-09-04:
@m-stein I refactored the nic driver as it did still use the old Nic driver interface. The lighttpd test works. The network_test_nic.inc seems deprecated thus I modified the network_test.inc in libports. The good news is that the server receives the packages, however it pauses for a few seconds every 399 packets so that the test case times out in the end.

chelmuth, 2015-09-29:
I was thinking about merging this line of work... am I right thinking that it drives the QEMU NIC?

@m-stein and @ValiValpas do the patches already apply to the spec changes from ed52d5a and could it be successfully rebased on current staging?

ValiValpas, 2015-09-29:
@chelmuth yes, it drives the QEMU NIC. I'm not sure about the spec changes though.

m-stein, 2015-09-30:
@chelmuth @valivalpas
I've rebased the current state onto staging without much trouble and removed the ``os/run/network_test_nic*`` files. The new state can be found in my branch ``1667_nic_support_Zynq_7000``.

When running ``lighttpd``, I get:
```
spawn qemu-system-arm -net user -redir tcp:5555::80 -nographic -serial mon:stdio -M xilinx-zynq-a9 -cpu cortex-a9 -m 256 -net nic,model=cadence_gem -kernel var/run/lighttpd/image.elf
Warning: nic cadence_gem.1 has no peer
kernel initialized
Genode 15.08-76-g518bfed
int main(): --- create local services ---
int main(): --- start init ---
int main(): transferred 1015 MB to init
int main(): --- init created, waiting for exit condition ---
[init -> lighttpd] void init_nic_dhcp(): init_nic_dhcp()
[init -> lighttpd] Using DHCP for interface configuration.
[init -> lighttpd] {anonymous}::Plugin::Plugin(): using the lwIP libc plugin
[init -> nic_drv] void Genode::Marvel_phy::phy_detection(): PHY address is not setup correctly 0
[init -> nic_drv] void Genode::Marvel_phy::phy_detection(): detecting phy address
[init -> nic_drv] void Genode::Marvel_phy::phy_detection(): PHY is not detected
[init -> nic_drv] void Genode::Marvel_phy::init(): The found phy has the id ffffffff
[init -> nic_drv] PHY reset timed out
[init -> nic_drv] Uncaught exception of type 'N6Genode10Marvel_phy23Phy_timeout_after_resetE' (use 'c++filt -t' to demangle)
[init -> nic_drv] abort called - thread: 'nic_ep'
```
When running ``network_test_lwip``, It seems to be the same problem:
```
spawn qemu-system-arm -net user,hostfwd=tcp::10000-:10000 -nographic -serial mon:stdio -M xilinx-zynq-a9 -cpu cortex-a9 -m 256 -net nic,model=cadence_gem -kernel var/run/network_test_lwip/image.elf
Warning: nic cadence_gem.1 has no peer
kernel initialized
Genode 15.08-77-g4ee40c8
int main(): --- create local services ---
int main(): --- start init ---
int main(): transferred 1016 MB to init
int main(): --- init created, waiting for exit condition ---
[init] parent provides
[init]   service "ROM"
[init]   service "RAM"
[init]   service "IRQ"
[init]   service "IO_MEM"
[init]   service "IO_PORT"
[init]   service "CAP"
[init]   service "PD"
[init]   service "RM"
[init]   service "CPU"
[init]   service "LOG"
[init]   service "SIGNAL"
[init] child "timer"
[init]   RAM quota:  765952
[init]   ELF binary: timer
[init]   priority:   0
[init]   provides service Timer
[init] child "test-ping_server_lwip"
[init]   RAM quota:  16494592
[init]   ELF binary: test-ping_server_lwip
[init]   priority:   0
[init] child "nic_drv"
[init]   RAM quota:  3911680
[init]   ELF binary: nic_drv
[init]   priority:   0
[init]   provides service Nic
[init] child "timer" announces service "Timer"
[init] child "nic_drv" announces service "Nic"
[init -> nic_drv] void Genode::Marvel_phy::phy_detection(): PHY address is not setup correctly 0
[init -> nic_drv] void Genode::Marvel_phy::phy_detection(): detecting phy address
[init -> nic_drv] void Genode::Marvel_phy::phy_detection(): PHY is not detected
[init -> nic_drv] void Genode::Marvel_phy::init(): The found phy has the id ffffffff
[init -> nic_drv] PHY reset timed out
[init -> nic_drv] Uncaught exception of type 'N6Genode10Marvel_phy23Phy_timeout_after_resetE' (use 'c++filt -t' to demangle)
[init -> nic_drv] abort called - thread: 'nic_ep'
```
I had a short look at the error location in the sources but it doesn't seem to be a trivial problem. My qemu-system-arm version is 2.0.0 (Debian 2.0.0+dfsg-2ubuntu1.15).

chelmuth, 2015-09-30:
I can confirm that it works with Qemu 2.3.1 on my notebook (and therefore should also work on buildbot). @ValiValpas if you're okay with @m-stein rebase I'll merge the patch series.

ValiValpas, 2015-10-02:
Yes, I'm okay with that.

chelmuth, 2015-10-02:
@m-stein expressed a vague offline reservation for an immediate merge and I'll wait for him to substantiate what may block us.

m-stein, 2015-10-06:
I'm on it.

chelmuth, 2015-10-06:
Okay, that's 156 additions and 135 deletions in 4 files. @ValiValpas could you please confirm @m-stein's changes?

ValiValpas, 2015-10-06:
@m-stein:
I tested your changes on my machine (Archlinux w/ qemu 2.3.0). Unfortunately, I get a pagefault in the nic_drv when running lighttpd:

```
spawn qemu-system-arm -net user -redir tcp:5555::80 -nographic -serial mon:stdio -M xilinx-zynq-a9 -cpu cortex-a9 -m 256 -net nic,model=cadence_gem -kernel var/run/lighttpd/image.elf
Warning: nic cadence_gem.1 has no peer
kernel initialized
Genode 15.05-305-g22b588c
int main(): --- create local services ---
int main(): --- start init ---
int main(): transferred 1015 MB to init
int main(): --- init created, waiting for exit condition ---
[init -> lighttpd] void init_nic_dhcp(): init_nic_dhcp()
[init -> lighttpd] Using DHCP for interface configuration.
[init -> lighttpd] {anonymous}::Plugin::Plugin(): using the lwIP libc plugin
[init -> nic_drv] void Genode::Marvel_phy::phy_detection(): Default phy address 0 is valid
[init -> nic_drv] void Genode::Marvel_phy::init(): The found phy has the id 01410cc2
[init -> nic_drv] int Genode::Marvel_phy::genphy_config_aneg(): Config changed
[init -> nic_drv] Could not open ROM session for module "config"
[init -> nic_drv] Could not obtain config file
[init -> lighttpd] void init_libc_lwip(): init_libc_lwip()
no RM attachment (faulter b28128 with IP 101364 attempts to read from address f000)
init -> nic_drv -> irq.36: unresolved pagefault at ip=101364 sp=e02feef0 fault address=f000
core -> pager_ep: cannot submit unknown signal context
[init -> lighttpd] read_rtc: rtc not configured, returning 0
[init -> lighttpd] pid_t getpid(): getpid not implemented
[init -> lighttpd] pid_t getpid(): getpid not implemented
[init -> lighttpd] libc_lwip: unsupported fcntl() request: 2
[init -> lighttpd] 1970-01-01 00:00:01: (/home/johanness/nosave/genode/contrib/lighttpd-cf80d75a9625c6d11ede200b7ee1cccffb16d5b0/src/app/lighttpd/src/network.c.427) can't set accept-filter 'httpready':  Protocol not available 
[init -> lighttpd] gid_t getgid(): getgid not implemented
[init -> lighttpd] uid_t getuid(): getuid not implemented
[init -> lighttpd] 1970-01-01 00:00:01: (/home/johanness/nosave/genode/contrib/lighttpd-cf80d75a9625c6d11ede200b7ee1cccffb16d5b0/src/app/lighttpd/src/log.c.166) server started 
[init -> nic_drv] Uncaught exception of type 'N6Genode20Tx_buffer_descriptor20Package_send_timeoutE' (use 'c++filt -t' to demangle)
[init -> nic_drv] abort called - thread: 'nic_ep'
```

Same with network_test_lwip:
```
spawn qemu-system-arm -net user,hostfwd=tcp::10000-:10000 -nographic -serial mon:stdio -M xilinx-zynq-a9 -cpu cortex-a9 -m 256 -net nic,model=cadence_gem -kernel var/run/network_test_lwip/image.elf
Warning: nic cadence_gem.1 has no peer
kernel initialized
Genode 15.05-305-g22b588c
int main(): --- create local services ---
int main(): --- start init ---
int main(): transferred 1016 MB to init
int main(): --- init created, waiting for exit condition ---
[init] parent provides
[init]   service "ROM"
[init]   service "RAM"
[init]   service "IRQ"
[init]   service "IO_MEM"
[init]   service "IO_PORT"
[init]   service "CAP"
[init]   service "PD"
[init]   service "RM"
[init]   service "CPU"
[init]   service "LOG"
[init]   service "SIGNAL"
[init] child "timer"
[init]   RAM quota:  765952
[init]   ELF binary: timer
[init]   priority:   0
[init]   provides service Timer
[init] child "test-ping_server_lwip"
[init]   RAM quota:  16494592
[init]   ELF binary: test-ping_server_lwip
[init]   priority:   0
[init] child "nic_drv"
[init]   RAM quota:  3911680
[init]   ELF binary: nic_drv
[init]   priority:   0
[init]   provides service Nic
[init] child "timer" announces service "Timer"
[init] child "nic_drv" announces service "Nic"
[init -> nic_drv] void Genode::Marvel_phy::phy_detection(): Default phy address 0 is valid
[init -> nic_drv] void Genode::Marvel_phy::init(): The found phy has the id 01410cc2
[init -> nic_drv] int Genode::Marvel_phy::genphy_config_aneg(): Config changed
[init -> nic_drv] Could not open ROM session for module "config"
[init -> nic_drv] Could not obtain config file
no RM attachment (faulter aa9128 with IP 101364 attempts to read from address f000)
init -> nic_drv -> irq.36: unresolved pagefault at ip=101364 sp=e02feef0 fault address=f000
core -> pager_ep: cannot submit unknown signal context
[init -> test-ping_server_lwip] Create new socket...
[init -> test-ping_server_lwip] Bind socket to 10000
[init -> test-ping_server_lwip] Listen on 0.0.0.0:10000...
[init -> test-ping_server_lwip] wait...
[init -> nic_drv] Uncaught exception of type 'N6Genode20Tx_buffer_descriptor20Package_send_timeoutE' (use 'c++filt -t' to demangle)
[init -> nic_drv] abort called - thread: 'nic_ep'
```

m-stein, 2015-10-07:
Sorry, I couldn't finish the review yesterday. I merely uploaded the commit to have an on-line backup.

m-stein, 2015-11-24:
@ValiValpas I'm sorry for the long silence. Because of my current schedule and the lack of testing hardware on my side, I would vote for skipping secondary issues for now and handle them in separate. I've rebased my [branch](https://github.com/m-stein/genode/tree/1667_nic_support_Zynq_7000) and removed my commits. Could you please test this state once more so we can be sure that it is working when it may enter the upcoming release?

ValiValpas, 2015-11-24:
@m-stein I'm missing the commit 082ac19 from my [branch](https://github.com/ValiValpas/genode/commits/1667_nic_support_Zynq_7000).

m-stein, 2015-11-25:
@ValiValpas I've update the [branch](https://github.com/ValiValpas/genode/commits/1667_nic_support_Zynq_7000). I had to manually rebase your commit to the version of the nic driver without my changes.

ValiValpas, 2015-11-25:
@m-stein alright. I ran some quick tests on qemu...it's working.

m-stein, 2015-11-25:
@ValiValpas Thank you for your fast feedback! :-)

chelmuth, 2015-11-25:
Looks good: `make run/lighttpd` survives `ab -c 10 -n 1000 http://localhost:5555/` and reports 33.30 requests per second ;-)

m-stein, 2015-11-25:
@chelmuth I've re-enabled the network tests for hw_zynq so I think we can merge it.

commit 2f1f89c4 (2015-11-25): re-enable network tests on hw_zynq

We disabled them recently for all platform w/o a NIC driver. Now
hw_zynq has got a NIC driver.

Fixes #1667
commit 3651992c (2015-11-26): re-enable network tests on hw_zynq

We disabled them recently for all platform w/o a NIC driver. Now
hw_zynq has got a NIC driver.

Fixes #1667
closed 2015-11-30 by m-stein