genodelabs/genode #1668: codezero: remove unsupported old kernel

skalk, 2015-09-03:
The development of the codezero kernel seems to be dead. Moreover the only platform supported is rather old. We currently have no ability to nightly test things on top of the kernel. We can only guarantee that things are compiling for the kernel, but nobody uses it at all. On the other side, every deeper change of core and other kernel dependent code leads to costs when having a lot of kernels/platforms. Therefore, I'd like to remove the codezero kernel support from Genode.

nfeske, 2015-09-03:
sniff... :-/ It seems like I am defeated. I can't help but have to agree with Stefan.

chelmuth, 2015-09-03:
Merged into staging - builder will be disabled in a second.

jayrulez, 2015-09-03:
Might Fiasco(not OC) and Pistachio be on the chopping block soon too?

nfeske, 2015-09-03:
Sooner or later, we will eventually remove them since they are essentially dead projects. If one of them should cause big trouble, e.g., when updating our tool chain, we probably won't invest much time to keep them supported.

But in contrast to Codezero, those kernels are still covered by our automated tests and we use them sporadically for debugging purposes. E.g., L4/Fiasco was very useful during our recent investigations with non-write-combined video memory, MTRRs, and the page-attribute table (PAT). Also, I tend to be a bit sentimental about the original L4 kernels. So keeping them around is nice and fuzzy. ;-)

commit 2d016929 (2015-09-03): fixup "codezero: remove support from Genode (fix #1668)"
commit ea01778d (2015-09-03): codezero: remove support from Genode (fix #1668)
commit acc46f70 (2015-09-09): codezero: remove support from Genode (fix #1668)
closed 2015-09-09 by skalk