genodelabs/genode #1669: foc_odroid_x2: Gpio Support.

alexygallar2, 2015-09-03:


alexygallar2, 2015-09-08:
This is the implementation for the basic configuration of GPIO pins for ODROID-x2. The interrupt handling is not implemented yet.

chelmuth, 2015-09-16:
@skalk same as with #1678. This contribution calls the platform _exynos4_ while #1678 calls it odroid_x2. Maybe that is okay but I want to double-check.

skalk, 2015-09-17:
@chelmuth : the same as with  #1678, the fixup commit is 19ff6bef01e7c53e0bfcba4f2b0f4e0869188a3a

@alexygallar2 : Thank you for your contribution! I've rebased your commit to current staging. Just a hint for future contributions: please have a look at our coding style (http://genode.org/documentation/developer-resources/coding_style), otherwise we've a lot of refactoring work to do on our side ;-).

chelmuth, 2015-09-17:
@alexygallar2 I merged 8d74e3f, 88bbc4b, and 1dabe56 into staging. Please review for rebase issues.

alexygallar2, 2015-09-17:
The gpio works well in ODROID-x2

commit 8d74e3fe (2015-09-17): foc_odroid_x2: basic configuration for gpio pins

Fixes #1669
commit de87ecfc (2015-09-30): foc_odroid_x2: basic configuration for gpio pins

Fixes #1669
closed 2015-09-30 by chelmuth