genodelabs/genode #1670: eglgears.run does not work

nfeske, 2015-09-03:
The eglgears.run script is broken. Besides some minor issue (a missing inclusion of the platform_drv.inc), the eglgears program fails with a segmentation fault on Linux. The backtrace looks as follows:

```
#0  Genode::Slab_block::slab_entry (this=this@entry=0x1, idx=idx@entry=0) at genode/repos/base/src/base/allocator/slab.cc:48
#1  0x50019d98 in Genode::Slab_block::slab_entry_idx (this=0x1, e=e@entry=0x12b76d8 <firsttime>)
    at genode/repos/base/src/base/allocator/slab.cc:54
#2  0x5001a006 in Genode::Slab_block::inc_avail (this=0x1, e=e@entry=0x12b76d8 <firsttime>)
    at genode/repos/base/src/base/allocator/slab.cc:84
#3  0x5001a284 in free (this=0x12b76d8 <firsttime>) at genode/repos/base/include/base/slab.h:148
#4  Genode::Slab::free (addr=addr@entry=0x12b76dc <do_clflush>) at genode/repos/base/src/base/allocator/slab.cc:268
#5  0x0140ab50 in Malloc::free (this=0x1453b20 <allocator()::placeholder>, 
    ptr=0x12b76e0 <Display::_get_connectors(native_display*, int*, int*)::conn_list>) at genode/repos/libports/src/lib/libc/malloc.cc:146
#6  0x0140ac92 in free (ptr=ptr@entry=0x12b76e0 <Display::_get_connectors(native_display*, int*, int*)::conn_list>)
    at genode/repos/libports/src/lib/libc/malloc.cc:196
#7  0x010c1d47 in egl_g3d_add_screens (drv=0x500d1428, dpy=0x500dc404)
    at genode/contrib/mesa-abe06d02df59ba0624e97c92a91ab2d9806d4889/src/lib/mesa/src/gallium/state_trackers/egl/common/egl_g3d.c:441
#8  egl_g3d_initialize (drv=0x500d1428, dpy=0x500dc404, major=0x400ffba8, minor=0x400ffbac)
    at genode/contrib/mesa-abe06d02df59ba0624e97c92a91ab2d9806d4889/src/lib/mesa/src/gallium/state_trackers/egl/common/egl_g3d.c:615
#9  0x010f5082 in eglInitialize (dpy=dpy@entry=0x500dc404, major=major@entry=0x400ffc28, minor=minor@entry=0x400ffc2c)
    at genode/contrib/mesa-abe06d02df59ba0624e97c92a91ab2d9806d4889/src/lib/mesa/src/egl/main/eglapi.c:280
#10 0x01000ce3 in main (argc=1, argv=0x50067400 <argv>) at genode/repos/libports/src/app/eglgears/eglgears.c:339
```

jameysharp, 2015-09-03:
In `repos/libports/src/lib/egl/driver.cc`, the implementation of `_get_connectors` returns the address of a statically allocated array. But `egl_g3d_add_screens` expects a malloc'd array, and passes it to `free` before returning. I'm surprised that the older toolchain tolerated this...

jameysharp, 2015-09-03:
I haven't written C++ in about 15 years, so I got some warnings I didn't understand when I tried to write this patch like I'd normally write C. But the following patch makes eglgears build in Genode 14.11 with the 15.05 toolchain. (I didn't know I could use a newer toolchain, but I did it by accident and it seems to have worked.)

```diff
diff --git a/repos/libports/src/lib/egl/driver.cc b/repos/libports/src/lib/egl/driver.cc
index 081bdf0..4d477c9 100644
--- a/repos/libports/src/lib/egl/driver.cc
+++ b/repos/libports/src/lib/egl/driver.cc
@@ -468,7 +468,9 @@ class Display : public native_display
                        int *num_connectors, int *num_crtcs)
        {
                static struct native_connector conn;
-               static const struct native_connector *conn_list[1] = { &conn };
+               const struct native_connector **conn_list;
+               conn_list = (const struct native_connector **) malloc(sizeof *conn_list);
+               conn_list[0] = &conn;
                printf("called, return 1 connector\n");
 
                if (num_connectors) *num_connectors = 1;
```

However, although 14.11's eglgears works in qemu with this patch, it still shows a black window on real hardware. 14.05 works though!

nfeske, 2015-09-04:
@jameysharp Thanks for investigating the issue! Your fix looks good to me.

>  I'm surprised that the older toolchain tolerated this...

I do not have a good explanation. Originally, our malloc implementation of the libc was just a wrapper around ```Genode::Heap```, which is robust against out-of-bounds arguments of the free function (```Genode::Heap``` just uses the pointer as a key for an AVL-tree lookup). However, for performance reasons, we later replaced the libc back-end allocator with a slab-based implementation (825dc783e6d5da338ac21b113927ac0f2ecafe4b). This implementation is not able to cope with out-of-range pointers. But the change was in 2012 already. Whereas this explains why eglgears worked back in version 10.08, I am a bit puzzled how it worked with version 14.05.

commit cc741d18 (2015-11-03): libports: fix eglgears example

Thanks to Jamey Sharp for the fix!

Fixes #1670
closed 2015-11-04 by nfeske