genodelabs/genode #1673: Apply SPEC approach consequently to all repositories

skalk, 2015-09-08:
Within the 'repos/base-hw' directory structure, we've test-wise applied the SPEC-approach to all aspects of the kernel/core. With SPEC-apporach I mean that the build system is totally controlled by SPEC variables. Moreover, we always distinguish at the various directory levels between generic and SPEC-related code, by having explicit 'spec' subdirectories containing the diverse SPEC-subdirectories. We should apply this approach to all Genode repositories, and thereby get rid of the quite overloaded 'platform' term. As 'platform_*' is just a specific form of aSPECt.

chelmuth, 2015-09-09:
Merged your work into staging and did some first fixups from `make drivers` on several platforms. Currently, I discovered only one open issue with *drivers/usb_net_stat* on several platforms.

skalk, 2015-09-10:
@chelmuth thanks for the first fixups, I've added another one to my staging branch, it should fix current build failures, and fix the run tests for Fiasco.OC

chelmuth, 2015-09-10:
Merged it just now \*cross fingers*

skalk, 2015-09-11:
I'm afraid we've to cross fingers again. @chelmuth commit ab33d282467c744f366a5da1dee43097f976179c is ready for staging.

chelmuth, 2015-09-11:
Okay 6c6d3db4c45519544c24b03d31222e98dac9dda2 and yours (0f08cd0f7c885757918d26dd6d630c9ed235b8e7) are now on staging \*cross fingers and thumbs*

commit 82adebde (2015-09-09): introduce 'spec' subdirectories to outline aspects

Instead of holding SPEC-variable dependent files and directories inline
within the repository structure, move them into 'spec' subdirectories
at the corresponding levels, e.g.:

  repos/base/include/spec
  repos/base/mk/spec
  repos/base/lib/mk/spec
  repos/base/src/core/spec
  ...

Moreover, this commit removes the 'platform' directories. That term was
used in an overloaded sense. All SPEC-relative 'platform' directories are
now named 'spec'. Other files, like for instance those related to the
kernel/architecture specific startup library, wh [... cut]
commit ed52d5a2 (2015-09-16): Introduce 'spec' subdirectories to outline aspects

Instead of holding SPEC-variable dependent files and directories inline
within the repository structure, move them into 'spec' subdirectories
at the corresponding levels, e.g.:

  repos/base/include/spec
  repos/base/mk/spec
  repos/base/lib/mk/spec
  repos/base/src/core/spec
  ...

Moreover, this commit removes the 'platform' directories. That term was
used in an overloaded sense. All SPEC-relative 'platform' directories are
now named 'spec'. Other files, like for instance those related to the
kernel/architecture specific startup library, wh [... cut]
closed 2015-09-16 by skalk