genodelabs/genode #1678: foc_odroid_x2: HDMI Support.

alexygallar2, 2015-09-09:
Support for driver of device HDMI on Odroid-x2 (Framebuffer)

alexygallar2, 2015-09-11:
Exynos5 framebuffer driver was adapted for it to be used also by the ODROID-x2, basically the changes are for the base addresses are taken from board_base.h and make it more generic. I tested with the scripts  framebuffer.run and demo.run using native resolution of 1920x1080 and worked fine on odroid-x2 .

chelmuth, 2015-09-16:
@skalk could you have a look at this contribution regarding the specs integration?

skalk, 2015-09-17:
@chelmuth: ok I've rebased the commit and produced a fixup commit regarding the specs integration, as well as some coding style fixes.

skalk, 2015-09-17:
the fixup commit is 19ff6bef01e7c53e0bfcba4f2b0f4e0869188a3a

chelmuth, 2015-09-17:
@alexygallar2 I merged 1027b8a, 50024cf, and 45d4106 into staging. Please review for rebasing issues.

alexygallar2, 2015-09-17:
@chelmuth Review the changes and I look very good. The tested changes in ODROID-x2 and everything works fine.

chelmuth, 2015-09-17:
@alexygallar2 Great, thanks for the test on real hardware.

commit 1027b8a8 (2015-09-17): exynos: generalize Exynos5 fb_drv for Exynos4/5

Fixes #1678
commit 49b3d520 (2015-09-30): exynos: generalize Exynos5 fb_drv for Exynos4/5

Fixes #1678
closed 2015-09-30 by chelmuth