genodelabs/genode #1686: init: propagate exit of children

nfeske, 2015-09-15:
In dynamic scenarios such as Turmvilla (#1552) subsystems should be able to gracefully exit. See also issue #1685. In many cases, subsystems are assembled out of several components using a nested instance of init. E.g., each of Turmvilla's "noux" and "linux" consist of several components such as nit_fb, terminal, ram_fs, etc. The lifetime of such a subsystem mostly depend on one of those components (i.e., the noux runtime or virtualbox). To enable CLI monitor (or the launcher) to automatically respond to an exiting noux instance, we need to propagate the exit of the noux component through the intermediate init runtime to the CLI monitor. To do that, I'd like to extend the configuration concept of init with an additional sub node for the ```<start>``` node:

```
<start name="noux">
  <exit propagate="yes"/>
...
</start>
```

If the ```propagate``` attribute is set to yes, the exit of the respective child will result in the exit of the entire init component, including all siblings of the exited component. The approach gives us the freedom to define even multiple children that may trigger the exit condition of init (by specifying an ```<exit>``` node at several ```<start>``` nodes. By using an ```<exit>``` node instead of an attribute, we have the freedom to enhance the mechanism in the future, e.g., by allowing the definition of an exit code to propagate. However, as I am not convinced of the usefulness of exit codes (as of now, we have them but they remain unused), I would simply forward the exit code from the child to init's parent.

chelmuth, 2015-09-16:
I've some remarks to the patch. First and most important, I fear it breaks all run scripts depending on `child "<child>" exited with exit value <value>` messages (e.g., signal.run) as it overloads `Child::exit()`, which prints the message. Additionally, I think we should document the feature - maybe in *os/doc/init.txt*?

Also, what I'm not sure about and I think you will enlighten me is how calling exit() really frees resources, i.e., how the siblings of the exiting child are halted.

chelmuth, 2015-09-16:
Okay, I now understand the exit() does not induce an exit of its siblings but only the propagation (according to the needs of  #1685). In this case, I think the commit message is misleading.

nfeske, 2015-09-16:
Could you please have a look at the new version?

chelmuth, 2015-09-16:
Just perfect and merged this second ;-)

commit cc1f3d73 (2015-09-16): init: propagate exit conditions of children

This patch extends the configuration concept of init with an additional
sub node for the <start> node:

<start name="noux">
  <exit propagate="yes"/>
  ...
</start>

If the 'propagate' attribute is set to "yes", the exit of the respective
child will appear to init's parent as the exit of the entire init
subsystem.

Fixes #1686
commit 0f052357 (2015-09-30): init: propagate exit conditions of children

This patch extends the configuration concept of init with an additional
sub node for the <start> node:

<start name="noux">
  <exit propagate="yes"/>
  ...
</start>

If the 'propagate' attribute is set to "yes", the exit of the respective
child will appear to init's parent as the exit of the entire init
subsystem.

Fixes #1686
closed 2015-09-30 by nfeske