genodelabs/genode #1687: VirtualBox: exit after shutdown

nfeske, 2015-09-15:
Currently, VirtualBox does never call ```Genode::env()->parent()->exit()```. For the Turmvilla scenario (#1552), I wish to automatically remove the rich OS subsystem once the VM is shut down. The issue is related to #1686 and #1685.

There is a function ```Shutdown()``` in the front end but - according to the log produced when triggering the shutdown - it is never called. How should I go to identify a suitable hook function inside VirtualBox to place the call to ```parent()->exit()```? @alex-ab Do you have a good idea?

alex-ab, 2015-09-15:
I presume that 'fireStateChangedEvent()' in  'repos/ports/src/virtualbox/frontend/console.cc' is called and the the state of 'MachineState_T a_state' turns to 'MachineState_PoweredOff'. That is probably the right trigger. To verify that just set the debug variable in console.cc and see whether it is actually called and what aState is telling us.

alex-ab, 2015-09-15:
```com::Shutdown()```  looks rather something related to the (X)COM protocol which we - for our port 	"short-circuited" - so maybe it is never called or is only called for fatal errors. Up to now it was never called because it is tagged as DUMMY.

alex-ab, 2015-09-18:
My guess was actually right. However there are some other issues letting the exit code not be called everytime E.g. one reason is that the pthreads of virtualbox destruct themselves which leads to some deadlocking issues ...

alex-ab, 2015-09-18:
Deadlocking within the virtualbox VMM process - i have to say.

chelmuth, 2015-09-18:
I thought this was fixed with #1505 ?

alex-ab, 2015-09-18:
I mean the threads which does the self-destruction succeed - however other alive threads become stuck - sometimes - especially the thread which does the VM shutdown ... Maybe some locks are still hold by the dying threads - don't know yet.

chelmuth, 2015-09-30:
@alex-ab could you please extract the issue-relevant commits in a issue branch?

alex-ab, 2015-09-30:
branch [issue_1687](https://github.com/alex-ab/genode/commits/issue_1687)

commit 9e278ae5 (2015-09-30): vbox: support shutdown of VMM

- send exit signal to parent
- avoid assertion in vbox posix backend

Fixes #1687
commit 6f3e9c12 (2015-10-06): vbox: support shutdown of VMM

- send exit signal to parent
- avoid assertion in vbox posix backend

Fixes #1687
closed 2015-10-06 by alex-ab