genodelabs/genode #1691: hw: null-pointer deference in IPC-path

skalk, 2015-09-16:
The HW kernel's Ipc_node class encapsulates communication partners and state of a thread, it is an implemented aspect of a thread. Normally, when a communication gets started, ended, or is canceled pointers to the communication partners change and its state gets refreshed. Unfortunately, in Ipc_node::_announced_request_cancelled the pointer to the "caller" is unset, but the state of the Ipc_node does not change. On the other hand, in member function Ipc_node::for_each_helper the "caller" pointer is dereferenced based on the current state without additionally validating the pointer's value.

To be honest, to me the Ipc_node states seem to be a bit over-engineered. For instance, I needed some assistance by another eye-pair to understand what PREPARE_AND_AWAIT_REPLY as state value is good for. In principle, there should be no need to track within the Ipc_node whether it is currently in preparation of a reply. Or whether it is in preparation of a reply and therefore was calling another instance and is waiting for that reply. When merging PREPARE_AND_AWAIT_REPLY into AWAIT_REPLY and PREPARE_REPLY into INACTIVE the IPC path gets a little more understandable and less error prone

m-stein, 2015-09-21:
I really like this simplification. The PREPARE_REPLY* states existed merely to ease the initial development of the IPC aspect. I have added a small (fix-up)[https://github.com/m-stein/genode/commit/da0f387ef7558f405f0ebdb702976c19007654a9].

skalk, 2015-09-21:
I'm ok with your fixup, thanks!

commit 519a0d68 (2015-09-21): Fixup "hw: simplify IPC node's state model (fix #1691)"

Minor style fix and log output on state errors.

Ref #1691
commit ff75423e (2015-09-21): hw: simplify IPC node's state model (fix #1691)

Merge the Ipc_node class' state PREPARE_AND_AWAIT_REPLY and AWAIT_REPLY,
as well as PREPARE_REPLY ans INACTIVE into one.
commit 37bae7bc (2015-09-30): hw: simplify IPC node's state model (fix #1691)

Merge the Ipc_node class' state PREPARE_AND_AWAIT_REPLY and AWAIT_REPLY,
as well as PREPARE_REPLY ans INACTIVE into one.
closed 2015-09-30 by skalk