genodelabs/genode #1693: Noux: local terminal-to-log service

ehmry, 2015-09-16:
I would like to see the option in Noux to run non-interactively and send all output to log, rather than open a duplex terminal session. Using the terminal-to-log component with Noux works fine for now, but I think using Noux for non-interactive tasks would justify a local solution. It would be a matter of a Noux configuration change, rather than spinning up another init and terminal component. The code that would be added to Noux should come under less than 100L and can be shared with the log_terminal. Sound reasonable?

nfeske, 2015-09-16:
Yes, it sounds perfectly reasonable.

Couldn't we use the VFS to our advantage? Similar to how we have the ```stdin``` and ```stdout``` attributes that point the libc to the standard input and output facility, we could do the same for noux. So whether to use the Terminal or the LOG session comes down to instantiating a ```<log/>``` or ```<terminal/``` in the VFS of the noux instance and specifying the path to the ```stdin``` and ```stdout``` config attributes. Noux wouldn't need to distinguish both cases and we'd be even more flexible than with your suggested solution. E.g., in principle, we could direct stdout to a block device, or feed data produced by the jitterentropy VFS plugin directly to stdin, or do both at the same time. The possibilities... ;-)

ehmry, 2015-09-16:
I will have to do some research on the unix stdio descriptors, but that does sound like the most flexible approach, I'll see what I can do.

ehmry, 2015-09-17:
I said it would take 100LOC but it came out to less than 50 :-)

ehmry, 2015-09-17:
Its configured like this, with a fallback to Terminal.
```xml
<start name="noux">
	<config stdin="/dev/null" stdout="/dev/log" stdin="/dev/terminal" />
</start>
```

nfeske, 2015-09-17:
Very nice. Have I already noted that I love the versatility of our VFS? And it seems to get even better. :+1:  Thanks @ehmry!

ehmry, 2015-09-17:
Uh, I've noticed that using <terminal/> nodes does not work as expected, but redirecting to a <log/> node is fine.

nfeske, 2015-09-17:
BTW, from my perspective, we could eventually even drop the fallback and remove the ```Terminal_io_channel```. This way, your stats would look even better. However, I am not 100% sure if the ```Vfs::Terminal_file_system``` is sufficient as a replacement right now. It would be worth a try. Also, we'd needed to adjust all noux-related run scripts.

ehmry, 2015-09-17:
I'm done pushing for now. If you attach jitterentropy to bash's stdin you get pretty funny results.

ehmry, 2015-09-17:
Yea, I wanted to maintain compatibility.

chelmuth, 2015-10-01:
@ehmry I merged d5dc25adef3d3b4cb9c yesterday but it seems to depend on some other work by you as the compilation fails
```
.../repos/ports/src/lib/libc_noux/plugin.cc: In member function ‘virtual int {anonymous}::Plugin::ftruncate(Libc::File_descriptor*, off_t)’:
.../repos/ports/src/lib/libc_noux/plugin.cc:1455:9: error: ‘FTRUNCATE_ERR_NO_SPACE’ is not a member of ‘Vfs::File_io_service’
    case Vfs::File_io_service::FTRUNCATE_ERR_NO_SPACE:  errno = ENOSPC; break;
         ^
make[3]: *** [plugin.o] Error 1
```
Could you please have a look at staging?

ehmry, 2015-10-01:
@chelmuth fixed at ed5f3da.

ehmry, 2015-10-01:
Or I could FTRUNCATE_ERR_NO_SPACE, but I think we need it eventually either way.

ehmry, 2015-10-01:
Right now noux prints the warning 'stdin VFS path not defined, connecting to Terminal session' which is a bit misleading because using a terminal node from the VFS will not work as well as Noux's internal Terminal_io_channel. So I think I should try and improve the Terminal node now.

nfeske, 2015-10-01:
The current show-stopper for using a terminal VFS node for noux is the inability of the VFS to issue asynchronous I/O. For noux, we need a way to block on read. The VFS terminal fs does not support that. The read call will just return with a len of 0. To fix that, we will need to change the VFS to perform read and write operations asynchronously. I think @chelmuth works in this direction.

ehmry, 2015-10-01:
@nfeske If I always return true from check_unblock at the terminal VFS node it works for me, could we just change how a block is detected in the noux channels?
https://github.com/genodelabs/genode/blob/master/repos/os/include/vfs/terminal_file_system.h#L94

ehmry, 2015-10-01:
551dc71 would seem to make it work.

ehmry, 2015-10-01:
Well, Vim is still not happy about it.

nfeske, 2015-10-01:
I am not 100% sure but doesn't this change let noux actively poll for terminal input all the time?

ehmry, 2015-10-01:
I'm not sure, I'll have to dig through the Noux some more.

commit d5dc25ad (2015-09-30): Noux: connect standard streams to VFS nodes

Fixes #1693
closed 2015-10-06 by chelmuth