genodelabs/genode #1696: Virtualbox: support clipboard guest addition feature

alex-ab, 2015-09-17:


alex-ab, 2015-09-17:
The current version works for a Linux Guest using text as transfer type (the other would be bitmap). One has to replace the mode attribute of ```<Clipboard mode="Disabled"/>``` in the respective VM configuration file *.vbox file to either ```Bidirectional```, ```HostToGuest``` or ```GuestToHost```.

For ```GuestToHost``` a Genode report will be generated whenever new data arrives from the Guest VM called "clipboard". For ```HostToGuest``` virtualbox will ask for a ROM called 'clipboard' and will respond to changes of the ROM. ```Bidirectional``` are both modes together enabled.

Will test with Windows guest the next days.

chelmuth, 2015-09-17:
Nice. I talked with @nfeske today and we agreed that UTF-8 would be the best text encoding. Is this also possible with available VirtualBox functions?

alex-ab, 2015-09-17:
In the commit I convert it from UTF16 (Guest) to iso-8859-1 (Genode) and the way back. I already use the VBox built-in convertation functions - there should be also one for UTF16toUTF8 - if we want to.

nfeske, 2015-09-18:
@chelmuth brought up UTF8 because it is more future-proof than latin-1. E.g., when a Chinese user wants to use the copy-paste mechanism between VMs, iso-8859-1 would not cut it but UTF8 would work well. Right now, the only component that somehow relies on iso-8859-1 is the terminal. But this, we could change quite easily.

alex-ab, 2015-09-18:
Clipboard in Windows 7 guest is now also functional.

nfeske, 2015-09-18:
Cool! :+1:

alex-ab, 2015-09-18:
The utf8 change is pending, but one can try the commit already.

alex-ab, 2015-09-23:
I removed the latin1 changes and now use the vbox utf8 functions. Also use the append function of the xml generator as implemented by @nfeske and reverted my change.

cproc, 2015-09-28:
Added a second fixup commit with more error checks and fixed string length calculation.

chelmuth, 2015-10-06:
Are there still open issues with this feature or may we generate a summarized feature commit?

alex-ab, 2015-10-06:
No - for me at least. @cproc ?

cproc, 2015-10-06:
I added a third fixup commit which uses 'append_sanitized()' instead of 'append()' and adds one more error check.

commit f206b92e (2015-10-06): vbox: enable clipboard support"

Fixes #1696
commit 5aec67d5 (2015-10-09): vbox: enable clipboard support"

Fixes #1696
closed 2015-10-09 by alex-ab