genodelabs/genode #1697: core: unsafe use of concurrently used allocators

skalk, 2015-09-17:
When debugging the bomb stress test on top of the HW kernel I observed page-faults within core during destruction of a Ram_session_component. The reason is the direct access to the dataspace slab allocator bypassing the Synchronized_allocator version of it. This "pattern" of using the unsafe interface of a Synchronized_allocator is used repeatedly within core, and may cause trouble in other scenarios.

To remedy the abuse of the "Synchronized_allocator::raw()" function, which returns the unsafe interface of the allocator, we should get rid of it at all.

chelmuth, 2015-09-22:
Merged it to staging but confirmation by @nfeske is pending.

nfeske, 2015-09-22:
Except for a minor detail I commented on inline, I like the patch very much. @skalk have you considered to let the ```Synced_range_allocator``` inherit the ```Synced_allocator``` and just use the ```operator ()``` for the range-allocator-specific functions? This way, you wouldn't need to implement the lock initialization twice.

skalk, 2015-09-23:
@nfeske: regarding the inheritance of Synced_range_allocator from Synced_allocator, well I need to be able to set the used lock from the outside via the constructor (needed by the Core_mem_allocator, because it uses one and the same lock for two different allocators), when Synced_range_allocator would extend the Synced_allocator, the base class API has to be changed to allow setting the lock too. For the time being, I didn't liked to open that pandora box, but leave the generic interface small, and hide the Synced_range_allocator in core's internal headers.

skalk, 2015-09-23:
@nfeske @chelmuth a fixup commit f5025218b8888a73d789529eafcdbd86d691a75d is ready for staging, it corrects the old header references the nightly build uncovered, and turns the lock pointer to a reference, like Norman observed correctly, thanks for reviewing!

commit 95a1fe21 (2015-09-22): safeguard the synchronized allocator template

* Move the Synced_interface from os -> base
* Align the naming of "synchronized" helpers to "Synced_*"
* Move Synced_range_allocator to core's private headers
* Remove the raw() and lock() members from Synced_allocator and
  Synced_range_allocator, and re-use the Synced_interface for them
* Make core's Mapped_mem_allocator a friend class of Synced_range_allocator
  to enable the needed "unsafe" access of its physical and virtual allocators

Fix #1697
closed 2015-09-30 by skalk