genodelabs/genode #1706: override libc access in Noux

ehmry, 2015-09-20:
My libc plugin access function is overriding the Noux implementation, the most obvious symptom of this is Vim complaining that all files are read-only because the ``access(..., W_OK)`` call is going thru the libc VFS rather than Noux syscalls.
https://github.com/genodelabs/genode/issues/1703

nfeske, 2015-09-22:
That makes sense. When looking at the code, I think that we (in the future) should consider moving most of the libc_noux code into a ```Vfs::Noux_file_system``` plugin and let the ```Libc::vfs_config``` function return something like ```<vfs><noux/></vfs>```. This way, we can eventually get rid of the original libc plugin mechanism and retain the VFS as the only plugin mechanism, saving one indirection.

ehmry, 2015-09-22:
Yea, this problem caught me off guard because at first I forget Noux had its own libc plugin.

commit 089bbf52 (2015-09-21): Override libc access in Noux plugin

Fixes #1706
commit 45bcb7f4 (2015-09-30): Override libc access in Noux plugin

Fixes #1706
closed 2015-09-30 by chelmuth