genodelabs/genode #1709: lwip: support for multi-threading is limited

chelmuth, 2015-09-22:
In a multi-threading environment the lwIP implementation ensures that the core functions are protected against concurrent access by a *tcpip* monitor thread. Unfortunately, this does not expand to complete reentrancy on the API level. The following description is taken from a recent [mailing-list posting](http://lists.nongnu.org/archive/html/lwip-users/2015-09/msg00030.html)

> The only lwIP constraint on netconn control block is that netconn functions are only thread safe against the core, they are not reentrant, so you cannot share the control block between thread without locking.

In other words: We can't use one and the same *socket* from different threads at the same time. As we do not ensure this invariance, you may see errors like the following
```
Assertion "tcpip_thread: invalid message" failed at .../src/lib/lwip/src/api/tcpip.c:195
```
or other strange behavior due to data corruption.

nfeske, 2017-12-13:
Since we are taking a completely different route with lwip now (using it as VFS plugin), I sense that this issue has become obsolete. @chelmuth may we close it?

chelmuth, 2017-12-14:
Agreed.

closed 2017-12-14 by chelmuth