genodelabs/genode #1710: RPI: only build the gpio driver if the user wants it

trimpim, 2015-09-22:
At the moment the GPIO driver is always built when the platform RIP is selected.

To keep the TCB as small as possible this should only be done when the user wants it.

alexygallar2, 2015-09-22:
Hi @trimpim 
We have fixed this compilation problem. You can see the solution at #1683. We think that if you don't want to compile the GPIO drv it can be configured at the run script to don't compile it. Why it should affect system's TCB?

chelmuth, 2015-09-23:
I agree with @alexygallar2 that what should be build is selected in the run script or command line by just listing targets that are relevant for the scenario. Further, the TCB of the scenario is not affected by built components but by the components integrated into the final image/system. So, if gpio_drv should not add to your TCB don't integrate it.

skalk, 2015-09-24:
@trimpim we use the SPEC variables to express what a platform, which means a hardware/kernel compilation, supports. It enables, or disables the compilation of certain drivers for certain platforms. For instance, if we have several different GPIO drivers, the SPEC variables can help to build only the right one for the chosen platform, when you type `make drivers/gpio`. However, what the platform actually supports is not chosen by the user, but given by the build directories chosen platform. In addition, the user can decide whether to build certain components or not, like @alexygallar2 and @chelmuth already explained. To sum it up, SPEC variables express what can be done, the user decides what should be done.
Are you fine with closing this issue?

closed 2015-09-30 by chelmuth