genodelabs/genode #1715: Block server backed by file system session

ehmry, 2015-09-25:


nfeske, 2015-09-28:
Have you considered to integrate this functionality into the VFS server? The VFS server could hand out block sessions in addition to file-system sessions if configured so. The provisioning of a block session could be configured via a policy node as follows:

```
<policy label="..." service="block"/>
```

If no service is specified, the VFS would hand out a file-system session. The announcement of the "Block" service could be configurable by a ```block``` attribute of the VFS' ```<config>``` node. The block service should be announced only when ```block``` is set to "yes".

The advantage of this approach would be that any VFS node could be exported as a block session, including ROM modules, RAM fs content, without the need for a separate VFS instance.

In the future we could enhance this concept to even cover other sessions. In particular, the input session comes in mind (thereby we could cover the use cases of the input merging, key mappings, key repeat configuration, touchscreen calibration by chaining VFS plugins that perform such input transformations).

BTW, it would be nice to have the comment fix for root/component.h in a separate commit.

ehmry, 2015-09-29:
I had considered using a VFS backend rather than the FS session, but I decided to go for something small and simple first.

Serving multiple services from the VFS server would be interesting, I think ROM sessions would be the easiest to do at first, because the VFS will already generate a dataspace for any given path.

ehmry, 2018-03-09:
@alex-ab check my fs_blk branch.

alex-ab, 2018-04-16:
@ehmry: I will try to have a look in the next days. Did you specify as block size 512, otherwise there are known issues #1676 ?

ehmry, 2018-04-16:
Yes, block size 512, and oddly the `test-rom_blk` test will verify the seoul-disc.raw image without any problems.

nfeske, 2020-07-03:
Fixed in master by the commits of issue #3781.

closed 2020-07-03 by nfeske