genodelabs/genode #1722: nova: interpret PF exception error code

chelmuth, 2015-09-28:
On NOVA, the UTCB field `qual[0]` is set to the exception error code by the kernel. We shall use the specified bits to check for error conditions, e.g., access violations. In my case ld.lib.so was infinitely looping in page-fault handling due to an unwanted text relocation, which ended up in a _write_ operation into the text segment.

chelmuth, 2015-09-28:
@alex-ab during the tests above I could not trigger any NX-related errors by instruction fetch. Does NOVA lack this feature currently?

alex-ab, 2015-09-29:
@chelmuth: yes. For me it start to work in principle after enabling nx support in [start.S](https://github.com/alex-ab/NOVA/blob/master/src/start.S#L74) by changing the value of the referenced line to 0x901. Howerver, still some more work is necessary to get it working. On the one hand in the kernel - by evaluating the execute bit of a Mem_crd and setting in the pagetable the NX bit of an entry). In Genode we already transfer this desired information during an rm_session->attach, but we don't evaluate and propagate the information in the generic [rm_session_component implementation](https://github.com/genodelabs/genode/blob/master/repos/base/src/core/rm_session_component.cc#L305) in core.

chelmuth, 2015-09-29:
@alex-ab thanks fpr the investigation. Would you mind to open a seperate feature issue?

alex-ab, 2015-10-07:
@chelmuth: please have a look whether the commit qualifies as solution. It just handles the write pagefault issue. The other cases like execute bit etc. handling I would address in #1723, since it also needs some minor changes to the generic base as also minor changes in all base-* platforms.

commit 427ffc89 (2015-10-08): nova: interpret write page fault correctly

Fixes #1722
commit 6c30bf26 (2015-10-09): nova: interpret write page fault correctly

Fixes #1722
closed 2015-10-09 by alex-ab