genodelabs/genode #1723: support of non-executable memory regions

alex-ab, 2015-09-29:
As pointed out in #1722: To support non-executable memory regions a principal extension in the generic Genode rm_session_component is required to propagate the information to the platform specific implementation. Furthermore potentially extensions/adjustments of the used base-* kernel are required.

alex-ab, 2017-10-18:
In principle I'm done. Please review. I will check some more places where we may have used attach instead of attach_executable.

skalk, 2017-10-18:
:+1: very cool alex-ab

chelmuth, 2017-10-18:
The patch series looks very good. Thanks @alex-ab!

nfeske, 2017-10-18:
That's great @alex-ab - especially that you got all the important kernels covered so quickly!

alex-ab, 2017-10-18:
In f59987df2530c7de822b09d769e21134261448b9  and 12417102bc24e237515cc487fd86c70daa789bec  I had to adapt mmap handling in libc to support executable memory (e.g. by Arora and Javascript). Please check.

chelmuth, 2017-10-18:
Looks good.

chelmuth, 2017-10-19:
The issue is marked fixed. Does this include 4333b4f?

alex-ab, 2017-10-19:
If you are strict, you may force me to open up a separate feature issue for 4333b4fa2236dc0306d368eadaca4ec2b1dd480a. The commit however is ready already.

chelmuth, 2017-10-19:
Merged the complete branch as 461a6fcff326f9621fe5937b8091236a8c30b863

commit b5170789 (2017-10-19): base: enable executable memory fault handling

Fixes #1723
commit 3331c803 (2017-10-20): base: enable executable memory fault handling

Fixes #1723
commit db329b02 (2017-11-01): base: enable executable memory fault handling

Fixes #1723
closed 2017-11-06 by chelmuth