genodelabs/genode #1726: vbox: make I/O port access less noisy

cnuke, 2015-09-29:
Whitelist a few more ports because Linux probes quite a few I/O ports at boot. Though it stands to reason if we need the PWRN message at all. Other OSes like OpenBSD probes all I/O ports in the ISA PnP range (0x0203 - 0x03ef) which slows down booting a lot.

chelmuth, 2015-09-29:
You may be right that we reached a state of development in which these diagnostic messages are not important anymore. What do you think about just adding a `verbose` flag which is set by default and white-lists all ports?

cnuke, 2015-09-29:
I would rather remove the code all together since it is merely a diagnostic feature that can be add easily again for debugging purposes. Than again there is no harm in having `if (verbose) PWRN(…);` and removing the white-listing.

chelmuth, 2015-09-30:
Yes, that sounds reasonable. Thanks for the adaption.

commit 2f8b3cfd (2015-09-30): vbox: make I/O port access less noisy

Instead of white-listing floppy controllers, certain SCSI controllers
as well as ISAPnP etc. pp. remove the diagnostics and print each access
if verbose is set.

Fixes #1726.
closed 2015-10-06 by cnuke