genodelabs/genode #1728: linux: recent kernels relocate ld.lib.so which is not supported

chelmuth, 2015-09-30:
In issue #1655, we identified that recent kernels scan the ld.lib.so ELF binary and detect the DYN type, which is required to allow dynamic binaries to be linked against the loader. Due to DYN, the kernel applies his address-space randomization magic and places ld.lib.so at an arbitrarily chosen location in the new process. If we force the binary to type EXEC via
```
/usr/local/genode-gcc/bin/genode-x86-elfedit --output-type exec ld.lib.so
```
the kernel complies with the load address in the ELF binary. So, one solution the the described issue could be to postprocess ld.lib.so for EXEC. But then, we have to find a solution to still provide the DYN variant for linking...

ssumpf, 2015-11-27:
@ValiValpas: 34b06c7 should fix issue #1655 without the need of an EXEC type ld.lib.so. Maybe you want to give it a try.

chelmuth, 2015-12-01:
@ssumpf I skipped 4e51425872 from your branch on merge as it seems to be a debugging vehicle.

chelmuth, 2015-12-01:
Also, on `make run/lwip_lx` on linux_x86 I get
```
var/libcache/base-common/base-common.lib.a(rm_session_mmap.o): In function `Genode::Native_config::context_area_virtual_base()':
.../repos/base-linux/include/base/native_types.h:137: undefined reference to `_context_area_start'
```

chelmuth, 2015-12-01:
The reason for the linker error is rooted in our lx_hybrid implementation, which uses a completely different thread implementation. Your patches introduce a generic dependency on the `_context_area_start` but the lx_hybrid implementation does not use the context area. It will also never use our linker, which also hints that the new dependency is purely artificial and we should revert it. What do you think about implementing `Native_config::context_area_virtual_base()` differently for hybrid and non-hybrid programs?

ssumpf, 2015-12-01:
'We originally introduced the so-called *lx_hybrid* base platform to ease the
building of Genode completely with the Linux host tools and linked to the host
libc. With this feature, we tried to accommodate the use of Genode as
component middleware on top of regular Linux distributions. For quite some
time, however, this feature remained unused. So we removed the lx_hybrid
special case and the corresponding 'always_hybrid' spec value.' -- release notes 15.11?

chelmuth, 2015-12-01:
We removed the always_hybrid spec, but cannot spare us to build some hybrid programs, e.g., lx_fs, audio_drv, nic_drv, and (most famously) fb_sdl.

ssumpf, 2015-12-01:
Ok, then they citation above seems to be misleading. Only 'always_hybrid' and not 'lx_hybrid' were removed. I will have a look into this.

chelmuth, 2015-12-10:
41b9f6bd0300a0b7f2940c9a0270e222eb4040d7 makes ld.so completely relocatable, which fixes this issue.

closed 2015-12-10 by chelmuth