genodelabs/genode #1729: base-hw for imx53 qsb with tz

daveti, 2015-09-30:
Hi there,

Is there a tutorial for running base-hw in the secure world and linux kernel in the normal world using imx53 qsb? There is no doc for imx53 in the code repo.

Thanks,
Dave

nfeske, 2015-10-02:
The TrustZone article at genode.org (http://genode.org/documentation/articles/trustzone) contains a link to the Genode branch that contains the i.MX53 SABRE tablet demo: https://github.com/skalk/genode/tree/i.MX53_tablet_demo The branch contains a README file that explains the scenario: https://github.com/skalk/genode/blob/i.MX53_tablet_demo/os/src/server/vmm/imx53/README Granted, the QSB is a bit different from the SABRE tablet, but the material hopefully still provides you with a good starting point.

chelmuth, 2015-10-02:
I think @nfeske answered the question and close the issue. @daveti please consider to refer for further discussions to our [mailing list](http://genode.org/community/mailing-lists) - we'd like to dedicate the issue tracker for tracking bugs, feature requests, or other issues and leave discussions to the mailing list.

closed 2015-10-02 by chelmuth