genodelabs/genode #1738: base: Dispatch signals in entry point

ssumpf, 2015-10-09:
Currently, when a signal arrives in the main thread, the signal dispatcher is retrieved  and called from the main thread, the dispatcher uses a proxy object that in turn sends an RPC to the entry point. This becomes a problem when the entry point destroys the dispatcher object, before the  `dispatch` function has been called by the main thread. Therefore, the main thread should simply send an RPC to the entry point upon signal arrival and the dispatching should be handled solely by the entry point.

nfeske, 2015-10-09:
Apart from the minor cosmetic detail (see my inline comment), the patch looks very good to me. Thanks @ssumpf for the rework!

chelmuth, 2015-10-09:
I'll add the fix to staging as soon as the mentioned function description is fixed/fixuped.

ssumpf, 2015-10-09:
@chelmuth: Do 09e48ef and 8e3d992 suffice?

chelmuth, 2015-10-09:
@ssumpf So shall it be.

ssumpf, 2015-10-09:
@chelmuth: So let the bot build

ssumpf, 2015-11-13:
@chelmuth: 04d0994 should implement the correct behaviour for base-hw

ssumpf, 2015-11-13:
@m-stein: Thanks for clarifying things.

m-stein, 2015-11-24:
@chelmuth @skalk Fixed.

commit e32c19ba (2015-11-16): base-hw: use signal context list for pending signals

'block_for_signal' and 'pending_signal' now set pending flag in signal context
in order to determine pending signal. The context list is also used by the
'Signal_receiver' during destruction.

Fixes #1738
commit 0c3dfbad (2015-11-27): base-hw: use signal context list for pending signals

'block_for_signal' and 'pending_signal' now set pending flag in signal context
in order to determine pending signal. The context list is also used by the
'Signal_receiver' during destruction.

Fixes #1738
closed 2015-11-30 by ssumpf