genodelabs/genode #1739: virtualbox real partition support

NobodyIII, 2015-10-24:
I am buying a laptop which seems likely to be compatible with Genode enough to handle a Turmvilla scenario. However, especially since it has an SSD, I would like to be able to have virtualbox run from a real partition rather than a VDI or some other type of disk image. Is there a simple way to add support for this feature, or will it require a fair amount of coding? Is there work planned in that direction?

cnuke, 2015-10-24:
VirtualBox already supports raw disk access and the guide [1] compiled by @cproc explains how to use it.

[1] https://github.com/cproc/genode/blob/vbox_vm_linux/doc/virtualbox_setup.txt

NobodyIII, 2015-10-24:
What about raw disk access under Genode, using a partition from part_blk?

cnuke, 2015-10-24:
That is working just fine - you only have to configure the VFS of VirtualBox to use a Block session and in addition route the Block session request to part_blk. You might take a look at the turmvilla scenario (#1552) to see how it is actually done.

NobodyIII, 2015-10-24:
How would this command work for Genode: VBoxManage internalcommands
createrawvmdk -filename home.vmdk -rawdisk /dev/sda2? Or is there something
different that will work? I'm still confused. Turmvilla isn't especially
well-documented.

cnuke, 2015-10-24:
It will not work on Genode. You have to prepare the .vmdk file you want to use beforehand on e.g. Linux. The same file later may then by used on Genode — that is how it is done in case of the turmvilla scenario and that one of the reasons why the VFS configuration mirrors the configuration of Linux, i.e. using `/dev/sda2`.

NobodyIII, 2015-10-25:
Thanks. It would be good if Turmvilla were better-documented.

chelmuth, 2015-10-26:
@NobodyIII If you are willing to document your experiences while enabling Turmvilla publicly, you may just add a link to #1552.

chelmuth, 2015-11-23:
The proposed feature is already in place - see #1552.

closed 2015-11-23 by chelmuth