genodelabs/genode #1741: hw_x86: detect and use all physical available memory

alex-ab, 2015-10-27:


alex-ab, 2015-10-28:
@reet- & @Kensan : with the patches hw_x86_64 is now able to detect and use all available physical memory of a machine by parsing the multiboot info structure provided by the bootloader. We need this to run on various machines and to start using more sophisticated workloads like running the noux tool_chain scenario. Can you please have a look for hw_x86_64_muen? The hw kernel now expects to find a multiboot info structure to scan for the available physical memory ranges. I suspect that they can be setup/provided by some muen subject configuration maybe (as you don't use a bootloader in the hw_muen case, right?) ? Thanks in advance.

Kensan, 2015-10-28:
Currently for us it would be simplest if the old behavior could be kept for hw_x86_64_muen in the short term. At a later stage a Muen tool could generate the Multiboot information for the Genode/hw_x86_64_muen subject during system integration time.

alex-ab, 2015-10-28:
Ok, I kept the original static physical ram detection for hw_muen and moved the dynamic physical ram detection to hw_x86_64 only platform_support code.

Kensan, 2015-10-28:
Thanks for following up on our suggestion. I gave 1d1be2d hw_x86_64_muen a spin on real hardware and everything looks fine from the Muen side of things.

commit df2ef9a8 (2015-10-29): hw_x86: discover all physical memory in core

Fixes #1741
commit 826c81ac (2015-11-04): hw_x86: discover all physical memory in core

Fixes #1741
closed 2015-11-04 by alex-ab