genodelabs/genode #1742: Intel-AMT-KVM Mouse and Keyboard are not recognized

C0A8647F, 2015-10-28:
It’s seems that the vendor-ID 8087:002c won’t initialize at os start-up. One reason could be that this usb device consists of more than one component. According Sebastian this happens at the usb interface initialization number 13.

------Debug-Log----------------------------------------------------------------------------
[init -> usb_drv] dev_warn: config 1 has an invalid interface number: 8 but max is 3
[init -> usb_drv] dev_warn: config 1 has no interface number 1
[init -> usb_drv] dev_warn: config 2 has an invalid interface number: 12 but max is 1
[init -> usb_drv] dev_warn: config 2 has an invalid interface number: 13 but max is 1
[init -> usb_drv] dev_warn: config 2 has an invalid interface number: 13 but max is 1
[init -> usb_drv] dev_warn: config 2 has no interface number 0
[init -> usb_drv] dev_warn: config 2 has no interface number 1
----------------------------------------------------------------------------------------------

chelmuth, 2015-10-28:
Please help me to clarify: The device is a vitual USB hub?

C0A8647F, 2015-10-28:
As far as I know Intel-AMT is hardware and therefore the connection couldn’t be virtual.

ssumpf, 2015-10-28:
@chelmuth: I suggested to enter this issue here and will a have look at it some time.

chelmuth, 2015-10-28:
@ssumpf might have known it ;-) I was just wondering why I did not experience this error before and therefore suspecting the (virtual) hardware to pop up only if AMT-KVM was configured.

chelmuth, 2016-07-20:
I tried to reproduce the issue on current staging but KVM pointing device and keyboard worked with `make run/usb_hid`
```
[init -> usb_drv] dev_info: new high-speed USB device number 8 using xhci_hcd
[init -> usb_drv] dev_info: I U
[init -> usb_drv] dev_info: input: USB HID v1.10 Keyboard [I U] on usb-dummy-11/input0
[init -> usb_drv] dev_info: I U
[init -> usb_drv] dev_info: input: USB HID v1.10 Mouse [I U] on usb-dummy-11/input1
[init -> usb_drv] dev_info: I U
[init -> usb_drv] dev_info: input: USB HID v1.10 Mouse [I U] on usb-dummy-11/input2
[init -> test-input] Input event type=MOTION 	code=0	rx=0	ry=0	ax=3185	ay=3440	key_cnt=0	
[init -> test-input] Input event type=MOTION 	code=0	rx=0	ry=0	ax=3140	ay=3399	key_cnt=0	
[init -> test-input] Input event type=MOTION 	code=0	rx=0	ry=0	ax=3072	ay=3358	key_cnt=0	
[init -> test-input] Input event type=MOTION 	code=0	rx=0	ry=0	ax=3003	ay=3317	key_cnt=0	
[init -> test-input] Input event type=MOTION 	code=0	rx=0	ry=0	ax=2912	ay=3276	key_cnt=0	
[init -> test-input] Input event type=PRESS  	code=35	rx=0	ry=0	ax=0	ay=0	key_cnt=1	KEY_H
[init -> test-input] Input event type=RELEASE	code=35	rx=0	ry=0	ax=0	ay=0	key_cnt=0	KEY_H
[init -> test-input] Input event type=PRESS  	code=30	rx=0	ry=0	ax=0	ay=0	key_cnt=1	KEY_A
[init -> test-input] Input event type=RELEASE	code=30	rx=0	ry=0	ax=0	ay=0	key_cnt=0	KEY_A
[init -> test-input] Input event type=PRESS  	code=38	rx=0	ry=0	ax=0	ay=0	key_cnt=1	KEY_L
[init -> test-input] Input event type=RELEASE	code=38	rx=0	ry=0	ax=0	ay=0	key_cnt=0	KEY_L
[init -> test-input] Input event type=PRESS  	code=38	rx=0	ry=0	ax=0	ay=0	key_cnt=1	KEY_L
[init -> test-input] Input event type=RELEASE	code=38	rx=0	ry=0	ax=0	ay=0	key_cnt=0	KEY_L
[init -> test-input] Input event type=PRESS  	code=24	rx=0	ry=0	ax=0	ay=0	key_cnt=1	KEY_O
[init -> test-input] Input event type=RELEASE	code=24	rx=0	ry=0	ax=0	ay=0	key_cnt=0	KEY_O
```

nfeske, 2016-11-30:
May we close this issue?

chelmuth, 2016-11-30:
AFAIK we finally fixed all remaining issues in summer with d4879f04a602f1affb00c6168574f549960f97f0.

closed 2016-11-30 by chelmuth