genodelabs/genode #1745: Auto ROM VFS plugin

ehmry, 2015-10-28:
At first I thought it would be nice if cli_monitor would request ROM sessions for subsystem configs not in the file system, and then I thought it would be nice if in noux I could read files that I knew were available as ROM without declaring each one in the config, then I realized I really just wanted another VFS plugin.

What you get is what you would expect, the following gives you access to any ROM provided by the parent: ```<dir name="rom"/> <auto-rom/> </dir>```

The ROM sessions are reference counted and persist until a call to ``sync(...)```, but I think I would prefer a ```void yield(void)``` method on the plugin interface to complement sync.

nfeske, 2015-10-29:
Very good idea! But have you considered to wrap the auto-ROM file system into a real plugin instead of adding it to the list of built-in file-system types? Please have a look at how the jitterentropy FS works. Putting the code into a dedicated shared library is a good way to avoid inflating the code complexity of the VFS by rarely used features. If it turns out that we use the auto-vfs very frequently, we can still decide to add it to the built-in file-system types to improve convenience. But right now, I would prefer a dedicated plugin.

ehmry, 2015-10-29:
@nfeske Thats fine with me, but do you like the name 'auto-rom' or 'any-rom'?

nfeske, 2015-10-29:
Of the two, I actually like `any-rom` better. Since the new file system offers a collection of ROM modules instead of only a single one, we could even go for the name `roms`.

cnuke, 2015-10-29:
I would rather choose `any-rom` because it is more explicit. `roms` only differs in the trailing `s` from `rom` which might lead to the wrong configuration.

nfeske, 2015-10-29:
@cnuke I see the point. But consider that a <rom> node always has at least a ```name``` attribute, whereas a ```<roms>``` node would not have any attributes. So personally, I think that this makes the chance to mix them up pretty low. That said, I am not against the name `any-rom` either.

ehmry, 2015-10-29:
On a side note, the reason why I want this plugin is that I have a server that provides ROM and File_system, but in actuality takes ROM filename or File_system root session arguments, rewrites that argument with result of a nix function applied to the argument, and forwards the new session request back to the parent. In this way cli_monitor subsystems can be generated on the fly, and something as simple as init can pull binaries straight from a nix package expression.

nfeske, 2015-10-29:
Sounds very cool! Does this component play the role of the "file-system filter" that we briefly discussed before (as it creates a local view onto a portion of the global store)?

ehmry, 2015-10-30:
@nfeske In theory yes, you could run an instance of the vfs server that would request its config thru the session argument rewriter, and then config that is produced would arrange a composite file system made up of store paths using root arguments.

ehmry, 2015-11-23:
I'm basing the issue branch I have for this off my branch for #1787, because either I
modify Attached_rom_session to take a label and filename argument or I modify the ROM 
framework below that to only require a label argument and pull the filename off of the 
end, if that makes sense.

ehmry, 2016-06-28:
I'm closing and will reopen at the world repository when the VFS constructors stabilize.

closed 2016-06-28 by ehmry