genodelabs/genode #1750: Unlink recursion

ehmry, 2015-10-29:
It is not clear at the VFS or File_system interfaces if unlink will recurse or fail on non-empty directories. I believe this is causing problems with libc and noux because under no conditions do either set errno to ENOTEMPTY.

POSIX does not do this, which I assume is keep operations in the kernel simple. An argument for recursive unlink would be that directory structures can be updated atomically at the block device, if there are VFS plugins that do that natively (I'm thinging of something like Venti).

I don't have a preference between the two for now, but we could always have it both ways with a recursive argument flag.

nfeske, 2015-10-29:
Is there actually a need to define the behavior? How about the following approach:

* We introduce an error for the not-empty case. So any file system that cannot deal with recursive unlink is able to reflect the problem to the caller.
* File systems that are able to perform unlink recursively just do so. They will just never return the error. This allows sophisticated file systems to sweep deep file hierarchies at once. Otherwise, the client would need to repeatedly unlink each file individually, which does not scale well.
* If the file-system user wants to make sure that a directory is empty because unlinking it, if can just look inside prior issuing the unlink operation.

ehmry, 2015-10-29:
Ok, I like this approach, I will add an error tomorrow.

ehmry, 2015-11-23:
I'm rebasing against master on the branch issue1750, the ram plugin is not recursive, but that can be fixed at the same time as #1780.

commit 992fa9f8 (2015-11-24): VFS, File_system: Not_empty

Unlink returns an error for non-empty directories when the backend does
not support recursive unlinking.

Fixes #1750
commit 1d92631e (2015-11-29): VFS, File_system: Not_empty

Unlink returns an error for non-empty directories when the backend does
not support recursive unlinking.

Fixes #1750
closed 2015-11-30 by chelmuth