genodelabs/genode #1752: Rump VFS plugin

ehmry, 2015-10-29:
As a continuation of the move from dedicated File_system servers to the more general VFS server #1648, this issue concerns accessing block-backed file systems directly from the VFS library using a Rump kernel.

I have a initial implementation that I have been using on my laptop for some time, but I've noticed in tests that multi-threading is a show stopper. It seems that the Rump kernel is confused about the so called implicit thread contexts when threads other than main call into the kernel. I have a dedicate sync thread that rump recognizes with an explicit context that I do not need to syncronize with the main thread, but using additional threads to call VFS cause kernel panics, even with a mutex just outside the kernel.

There is also the issue of whether the VFS should perform recursive unlinking or defer to the BSD behaviour #1750.

nfeske, 2015-10-29:
This issue will go away automatically as we will ultimately remove the thread-safety of the VFS. This will then also allow us to remove the ugly locking stuff from the VFS code.

@chelmuth is currently in the process of redesigning the control flows within the libc to make sure that the libc calls the VFS by a single thread only. The only other remaining offender would be noux as it relies on multiple threads - one child thread for each process. The logical consequence is to change noux to use the same entrypoint to serve all processes. But to make this happen, we need to redesign the noux session interface to work asynchronously (right now, the processes block on the RPC interface). Alternatively, we could try to wrap the VFS in a ```Synced_interface``` inside noux.

Does your current solution (VFS + VFS-rump plugin) already works as a replacement for rump_fs? If yes, has @cnuke any objections against replacing it?

ehmry, 2015-10-29:
I've not noticed any runtime difference between them, and the two threads of VFS server don't interfere with each other for me.

cnuke, 2015-10-29:
If it works as advised I have not objections — go for it.

ehmry, 2015-10-30:
There is a resource leak somewhere that exhausts the allocator backend for rump and causes a failed allocation loop. I can trigger it consistently when I make a copy of the gcc noux-pkg directory. I've looked hard at the VFS plugin and watched the calls that lead up to the exhaustion and I think its something that is happening in the rump kernel, not the plugin, but I still have to test this with rump_fs.

ehmry, 2015-11-04:
I solved my issue with Rump allocation, I changed the allocator so it enforces a quota that is read from config at construction. This value is also reported to kernel thru the RUMP_MEMLIMIT parameter. A top level config node should be fine since there is at most one kernel per-component.

I'm not sure why this fixed it, I still see occasional allocation failures but the pagedaemon must be freeing memory, it could be a difference between the reported RUMP_MEMLIMIT and the failure limit. Either way this is an improvement because the current allocator will request up to 128 1MB dataspaces that can't be reclaimed from the kernel.

ehmry, 2015-11-04:
I add the verbosity option for sake of demonstrations :-)

ehmry, 2015-11-04:
I have it on good authority now that the pagedaemon does not react to allocation failures as the Rump paper says, but on RUMP_MEMLIMIT.

ehmry, 2015-11-30:
rebase to 15.11

nfeske, 2015-11-30:
That was quick! :-)

ehmry, 2016-03-01:
Rump_fs seems to be passing up **.** and **..** during directory reads, libc will drop these, but the VFS library will not, but the plugin strips these.

nfeske, 2016-04-11:
@ehmry Do you regard the rump VFS plugin as ready for staging?

ehmry, 2016-04-11:
No, I've noticed after using it for quite a while that there is a issue with the sync thread. I should be able to get that fixed in a few days.

ehmry, 2016-04-11:
I will say its ready now. The problem I was having was between the thread that synced the rump kernel and block device and the VFS server thread. For now I've put a lock that synchronizes the the two, but in the future this could be fixed by handing the file system or the file system factory an entrypoint for the server thread, as mentioned in #1934.

ehmry, 2016-04-11:
Because this issue is so old I can rebase against master if it makes things easier.

chelmuth, 2016-04-11:
I'd appreciate the rebase, thanks in advance.

ehmry, 2016-04-11:
f3616a134f1efafce6c8857c4a2943a273ace373 and a677c055a43e8551bbbb445af41f9452613ab943 build against master, 5d53d0e1e4263ea0ec75c4dd1835561fdb893eeb needs the handle allocator changes that are on staging.

ehmry, 2016-04-13:
I replaced the lock with a mutex, but I get an intermittent fault during the read part of ``run/libc_vfs_ext2``, and its always at address 0x40.

ehmry, 2016-05-21:
This fault at 0x40 seems to be related to latency in the block session, I don't have problems running the scenario on linux, but I do in QEMU, and I didn't have problems on an SSD, but I do on platters.

@ssumpf any ideas?

commit d04c603a (2017-01-09): Rump kernel VFS plugin

Tests at libc_vfs_ext2, libc_vfs_fs_ext2.

Fix #1752
Ref #2224
closed 2017-01-13 by nfeske