genodelabs/genode #1753: init: fix uninitialized variables

cproc, 2015-10-29:
The 'cpu_quota_pc' and 'constrain_phys' member variables of the 'Init::Child::Resources' structure remain uninitialized if they are not defined in the config file.

cproc, 2015-10-29:
Commit e800346 initializes 'cpu_quota_pc' with 0, which is the value it had in its uninitialized state most of the time. But I'm not really sure if this is the correct value.

chelmuth, 2015-10-30:
c9272937e76541a876749aca7350cacef560edee applied the following change in child.h
```patch
 			Read_quota(Genode::Xml_node start_node,
 			           Genode::size_t & ram_quota,
-			           Genode::size_t & cpu_quota,
+			           Genode::size_t & cpu_quota_pc,
 			           bool           & constrain_phys)
 			{
 				Genode::Number_of_bytes ram_bytes = 0;
-				Genode::size_t          cpu_percent = 0;
 
 				try {
```
@m-stein would it still be correct to set `cpu_quota_pc = 0`?

chelmuth, 2015-11-23:
@m-stein do you have any reservations to apply e8003466622fe0e4a480b0fa15e849c8ecca6114 fro m the background of my comment above?

m-stein, 2015-11-24:
@chelmuth @cproc Looks fine. Thank you for the fix!

commit 3b95eea1 (2015-11-24): init: fix uninitialized variables

Fixes #1753
commit 25643fa0 (2015-11-29): init: fix uninitialized variables

Fixes #1753
closed 2015-11-30 by cproc