genodelabs/genode #1756: Implement sigprocmask

ehmry, 2015-10-30:
Only because the DUMMY macro is trashing my log output.

cnuke, 2015-10-30:
I would rather implement `sigprocmask` locally then adding it to the libc. In any case the implementation in the libc has to be weak so that other plugins may override the symbol.

chelmuth, 2015-10-30:
Why should the implementation be weak? I can't image how plugins may effectively implement signals on their own - remember how complex select() grew. We rather have to implement signals in the libc and enable backends, e.g., the VFS plugin, to notify.

cnuke, 2015-10-30:
I agree that it should be handled by the libc itself eventually but currently there is `libc_noux` that depends on overriding `sigprocmask()`.

ehmry, 2016-01-08:
I made it weak.

chelmuth, 2016-01-08:
I noticed and it seems the way to go for now ;-)

commit 527d7e6e (2016-01-08): Libc: sigprocmask reports all signals blocked

Fixes #1756
commit b9c234a3 (2016-01-13): Libc: sigprocmask reports all signals blocked

Fixes #1756
closed 2016-01-13 by chelmuth