genodelabs/genode #1763: vfs: report file system

nfeske, 2015-11-04:
It would be nice to supplement the set of built-in VFS file systems with a new "report file system" that presents a write-only file. Once the file is written to and closed, its content should be submitted to a `Report::Connection`. With such a file system in place, we could easily integrate clipboard support into the noux environment of the Turmvilla scenario (#1552): Mounting the clipboard's ROM as a `<rom>` node and the facility to post clipboard content as a `<report>` node.

ehmry, 2017-04-29:
I have this implemented at 346c9fed76bbf67668d7f18be21f0c675529980c.

fbfbb1bae75729b3303bfa33443c7f7273f141f9 is to make sure that the test works, and 2ac4ce760b515a6253c0a1ea12cabd4c9f1489a9 makes ``sync`` available from ``Vfs_handle`` which has been an annoyance of mine for some time.

Right now I'm playing around with the idea of injecting subsystems into ``init`` configurations with ``Report`` and I'd like to prototype with Bash scripts.

NobodyIII, 2018-03-05:
Is there any issue keeping this from being merged?

chelmuth, 2018-03-14:
Good question. @ehmry are you using this feature regulary resp. do you have a minimal up-to-date patch?

ehmry, 2018-03-14:
No, I have not been using this, I wasn't sure if I was the one who implemented it or not, I can update and test in the next day or so.

chelmuth, 2018-05-15:
This came to my mind today as we were discussing about progress reports from ported (Noux) tools.

chelmuth, 2019-03-14:
Any plans to finally merge this feature?

ehmry, 2019-03-15:
Shall I rebase and update?

nfeske, 2019-03-15:
I don't know. Even though I was the one who originally requested the feature, I haven't actually had a practical use case for it so far.

NobodyIII, 2019-03-15:
It seems like it would simplify Sculpt a bit.

nfeske, 2019-03-17:
@NobodyIII Could you possibly be more specific? Am I unable to see the obvious?

chelmuth, 2022-08-31:
This still seams a tempting feature but is stale since 3 years. Therefore, let's close it for now.

closed 2022-08-31 by chelmuth