genodelabs/genode #1764: port Intel i915 Linux driver

skalk, 2015-11-04:


nfeske, 2015-11-04:
I have just integrated the new driver into the Turmvilla scenario and it works! I am excited! Still, a few refinements are needed to make it usable in practice:

* I have connected an 1920x1200 display to the x201, which has an internal 1280x800 display. When booting with the external display connected (via display port), the external display shows a resolution of 1920x1200 but only the upper part is used by nitpicker. It seems that the driver reports a framebuffer size of 1920x800 to nitpicker. When plugging in the external monitor after having booted, the external display shows a picture of 1024x768 (reported by the on-screen menu). Ideally, the driver should always pick the highest available resolution of all connected displays (in this case 1920x1200) and report it (and subsequent mode changes when displays get connected/unconnected) to nitpicker.
* The driver lacks the buffered mode, which the VESA driver has. Consequently, there are visible artifacts whenever we use alpha channels. It would be great if the driver implemented the `buffered` option.
* The driver does not deliver sync signals. However, on the turmvilla branch, nitpicker and the other GUI components use those signals for synchronization. As an interim solution, we could use the timer to provide periodic signals as I do in the commit https://github.com/nfeske/genode/commit/45d4e44055a50993131e0695db72d4133520040c. This is what I have done now. In the future, those signals should be triggered by the vblank.
* I had to disable the IOMMU for now, which is a known issue. I think @alex-ab is investigating it.

alex-ab, 2015-11-05:
Nice !  Is it supposed to work for 32bit Genode hosts?

skalk, 2015-11-05:
@nfeske fixup 74f680a0c28a877a63bc8b87d807743b83da6500 solves the first described problem

nfeske, 2015-11-05:
Works superb! BTW, I have pushed my intel-fb-enhanced turmvilla branch to https://github.com/nfeske/genode/commits/turmvilla_intel_fb

chelmuth, 2015-11-10:
Great work, but I'd like to defer adding the `fixed` label until you both agree on a merge.

alex-ab, 2015-11-11:
Branch [issue_1764](https://github.com/alex-ab/genode/commits/issue_1764) contains necessary commits to use the intel_fb driver with iommu enabled. I have several machines working now flawlessly, beside X201 and X250. For the X250 - the dmar unit reports that it is capable of using superpages, but the hardware faults - saying it does not like superpages (035111c94cf06e31a8a172ecd28ae507f98e0a3d is a immediate workaround if needed).

alex-ab, 2015-11-12:
0ca26d5ea314182d871869714d35f4c7badbf13a fixes for me the Lenovo X201 issues - we seem not to need the functionality which causes trouble - so I disabled it - please test it , esp @nfeske

cnuke, 2015-11-13:
We might want to add `REQUIRES = vesa` (or instead some other value) to the target.mk. Otherwise the driver gets build on e.g. base-linux.

chelmuth, 2015-11-13:
The missing dependency might be `REQUIRES = pci`, right?

alex-ab, 2015-11-13:
@chelmuth: please add also 92f619be6fd8286ac38f75c8631bfcddcbac22fa to staging, it fixes usb issues

skalk, 2015-11-25:
@chelmuth @nfeske : my issue branch https://github.com/skalk/genode/tree/issue%231764 contains all relevant commits for the intel driver, including reports and dynamic (re-)configuration.

commit 2ee0c524 (2015-11-25): dde_linux: KMS-based Intel framebuffer driver (II)

* enable i915 driver from Linux 3.14.5
* tested for generation 5 till 8 GPUs

Fix #1764
commit 4889eb30 (2015-11-25): fixup "dde_linux: KMS-based Intel framebuffer driver (II)"

* enable i915 driver from Linux 3.14.5
* tested for generation 5 till 8 GPUs

Fix #1764
commit e74ee60f (2015-11-26): dde_linux: KMS-based Intel framebuffer driver (II)

* enable i915 driver from Linux 3.14.5
* tested for generation 5 till 8 GPUs

The driver can be configured at run-time via the config ROM. Every
connector of the graphic card can be configured separately using the
following syntax

  <config>
    <connector name="LVDS-11" width="1280" height="800" enabled="true"/>
  </config>

Also, when enabled within the intel framebuffer driver configuration like
the following

  <config buffered="yes"/>

a simple ram dataspace is propagated to the client and the driver
itselfs copies from that buffer to t [... cut]
closed 2015-11-30 by skalk