genodelabs/genode #1766: Init: route sessions with partial label matches

ehmry, 2015-11-05:
I propose an ammendment to 9c13933ee75c1aa2b80d74c7d8a7f7f0430ba40e.

It would make some things easier for me if init could route requests using a partial match method similar to the one used for session policies.

This way one could do something like this:
```xml
<route>
 <service name="ROM" label="init ->"> <child name="fs_rom"/> </service>
 <service name="ROM" label="init">    <parent/> </service>
</route>
```

This is a a little different from policy matching in that the first pattern match returns, rather than the longest match of every label pattern specified.

nfeske, 2015-11-05:
I considered this approach. But it has the disadvantage that it becomes impossible to uniquely express a fixed label. E.g., in the example above, it becomes impossible to express a route that matches "init" but not "init.2".

In the longer term, we might consider introducing a simple pattern-matching mechanism that would also allow to match labels that have a certain suffix. For example, in the turmvilla scenario, I need to route all requests for `hypervisor_info_page` to core, regardless of the subsystem where the request originates. Hence, I'd like to express something like "*hypervisor_into_page". I would not go as far as using regular expressions though, because I am a bit afraid of the code complexity and stack usage that comes with it.

chelmuth, 2015-11-05:
I endorse a approach like [globbing](https://en.wikipedia.org/wiki/Glob_%28programming%29#Syntax) and think it would be best if label matching does not differ between init and general policy config in other nodes.

ehmry, 2015-11-05:
Globbing would be better. I think most libc's come with a globber, I can look for one.

ehmry, 2015-11-05:
Or something like ``<if-arg key="label" lmatch="init.2" rmatch="init.2">`` would be pretty simple. I would still like to see something like ``<if-arg>`` stay around because I like to match on arbitrary arguments, like ``<if-arg key="filename" value="hypervisor_info_page"/>``.

nfeske, 2015-11-05:
The `filename` session argument will be dropped sooner or later. The motivation to use solely the label for routing sessions and selecting policies is to keep the configuration manageable. E.g. let us assume that we add session-routing arguments to CLI monitor. With only the label as policy-relevant attribute, we could use a syntax like the following:
```
start noux --route File_system(tmp):tmp_fs(noux_tmp)
```
This would tell CLI monitor to route the File_system session with the label "tmp" to the child tmp_fs, presenting "noux_tmp" as session label to the server. Now imagine to offer the freedom of <if-arg> at this level. It would become a syntactic nightmare.

Hence, we tried to cut down the use of policy-relevant session parameters, except for the label. Interestingly, we could eliminate all of them. The filename argument of ROM sessions are really a relic. ROM servers should take the last portion of the label as module name.

nfeske, 2015-11-05:
As for the "globbing", how about introducing two new attributes `label_prefix` and `label_suffix` in addition to the current `label`? This would be fairly intuitive and easy to implement. I agree with @chelmuth that we should use the same mechanism for policy selection and session routing.

chelmuth, 2015-11-05:
:+1: from me for supporting label prefix and suffix without a new syntax in the attribute strings. The `label` attribute is kept for exact matches then, right?

ehmry, 2015-11-05:
I like it.

nfeske, 2015-11-05:
@chelmuth Yes, that was my intention.

ehmry, 2015-11-05:
I'll start on this now, because I will use it right away.

ehmry, 2015-11-09:
I implemented matching on label, label_prefix, and label_suffix. This applies anywhere session_policy is used so it may break a few more things than the fs_log.

ehmry, 2015-11-18:
I don't like label="" as a wildcard and I think it should be dropped. The problem is we can update the ones in the Genode source but this would be a breaking change for external configurations.

nfeske, 2015-11-19:
I agree. The label attribute should be a perfect match, not a wildcard. Of course, this requires us to modify existing configurations but it is better to implement such a change sooner rather than later. 

From a Genoder-user point of view, the change is pretty simple: Just replace "label" by "label_prefix" and you are fine. Since we tighten the semantics of the label attribute, this change will not weaken the security of existing policies. In the worst case, a server will reject sessions that formerly worked, but not vice versa.

In short, please go ahead with changing the meaning of the label attribute! :-) The adaptation of all the existing run scripts should be done via a separate commit.

ehmry, 2015-11-23:
Rebased against master at branch issue1766.

nfeske, 2015-11-24:
@ehmry: @chelmuth and me reviewed your label-matching improvement today. While doing that, we recognized that I missed to specify my desired matching semantics precise enough. I think that the three attributes are not equally significant. In particular, the number of matching prefix characters should not be weighted against the number of suffix characters. Instead, I want to make sure that there is a strict order of their significance. A perfectly matching `label` is strongest. Only if two nodes have an equally strong label, the `prefix` should be considered. Only if the prefixes are equally strong, the suffixes should be considered. We found it difficult to model these rules with a single (and also tangible) score value.

The rationale behind this thinking is that the leading part of the label is more "trustworthy" than the later parts as the prefixes are successively added by the chain of parents.

The commit https://github.com/nfeske/genode/commit/e24ad8342e07e7663ac7c9cbf9f236c1842643b7 implements the idea. Do you approve?

ehmry, 2015-11-24:
Yes, that comes out easier to read and its less amigious when scoring in both directions.

nfeske, 2015-11-25:
I merged the new version and your follow-up commit to staging. While reviewing the latter, I made some minor additional adjustments (the fixup commit).

ehmry, 2015-11-25:
``fixup "access XML types as Strings"`` is only significant to policy changes at report_rom, which is something that can be dealt with in another issue. Its actually only on this branch because it marks for_each_node as const.

nfeske, 2015-11-25:
Why have to opted to call the new type `Xml_type` instead of `Xml_node::Type`? I would prefer to keep the type local to the scope of `Xml_node`.

ehmry, 2015-11-25:
Could Xml_node::Type be returned from both Xml_node and Xml_attribute?

nfeske, 2015-11-25:
Oh, now I see. Actually, calling the name of an attribute "type" was probably a mistake on my side in the first place. I actually forgot that the `Xml_attribute::type` function even exists. ;-) In practice, we usually talk about "attribute names and values" and "node types". Hence, here is what I would like best:

```
class Xml_attribute
{
  typedef String<64> Name;
  Name name() const;
  ...
};

class Xml_node
{
  typedef String<64> Type;
  Type type() const;
  ...
};
```

ehmry, 2015-11-25:
Yes, thats nice and clean.

ehmry, 2015-11-27:
I recommend cherry-picking ``server/fs_log: merge labels using 'label_prefix'``, otherwise the log merging feature is lost at fs_log.

Anything else on my issue branch can go with different issue.

ehmry, 2015-11-30:
Closed at 3f6ff39d76bf0abb6ba7cd22ce37261894dd2497

closed 2015-11-30 by ehmry