genodelabs/genode #1775: File_system support for SEEK_END

ehmry, 2015-11-12:
Appending to a file with the File_system service is problematic because if two clients try to append to the same file simultaneously, they may write to the same file position, even if they both stat the file before writting. This is because clients have no guarantee that there are no pending write packets in a queue at the time of a stat.

ehmry, 2015-11-12:
The other solution would be to truncate before writing, but that would be a stat and a truncate before a write.

ehmry, 2015-11-18:
SEEK_END isn't going to work, because its a macro in libc, so I propose SEEK_TAIL.

chelmuth, 2016-04-25:
@ehmry I'd like to merge SEEK_TAIL support. Could you please rebase on current master (or even better staging ;-)

ehmry, 2016-04-25:
Ok, I have changes to the session interface, ram_fs, and vfs at ccf846d6a557aa1aa1e55c0184a00a506687cfda

chelmuth, 2016-04-26:
After look at ccf846d I think it might be good to add more documentation to SEEK_TAIL describing its semantics esp. with respect to read and write.

ehmry, 2016-04-26:
d82af00e7f503c455290c8d09fc4e43e5a5580ea is a replacement for ccf846d6a557aa1aa1e55c0184a00a506687cfda with documentation.

I'm not convinced SEEK_TAIL is the best name, tail can be an overloaded concept.

ehmry, 2016-04-27:
I have the corrected the read and write positioning in e59bff468a2f9195d59c9e92567e2de5cf6b0565, and implemented rump_fs support.

chelmuth, 2016-05-04:
@ehmry I merged your latest patch. Please have a look at my attempt 20a2adb to document the semantics of SEEK_TAIL in read and write. Do you approve?

ehmry, 2016-05-04:
Yes, looks good.

closed 2016-05-09 by chelmuth

ehmry, 2016-05-25:
@chelmuth 1484d05e7890716525e5ec41580be629dd92bfb0 should be applied to master.

commit 51facce2 (2016-05-04): Amend File_system session with SEEK_TAIL support

Used to read or write from the end of a file when multiple packets may
be in transit.

Supported by ram_fs, rump_fs, and vfs servers.

Fixes #1775