genodelabs/genode #1777: fs_log: heap partitioning

ehmry, 2015-11-12:
With SEEK_END #1775 fs_log no longer needs to track file write seeks between sessions, so no shared state is needed.

ehmry, 2016-04-25:
This can wait until SEEK_TAIL hits master.

ehmry, 2016-05-06:
I consolidated all the changes I've made to ``fs_log`` to 73465cbbfa453320a341ce5eaae89a0680345b21. 7fc4e1441f921a7312ca3cf28da018da60a7eee3 is a utility that I've used in other components and seems to work well for mapping session hierarchy to the file system.

nfeske, 2016-05-09:
The `path_from_label` function is a good idea. But why don't you let the function return the result? I would find it more intuitive to do something like:
```
typedef Genode::Path<128> Path;
Path const path = path_from_label<Path>(label);
```
Granted, it may be merely a matter of taste. But this way enables us to make `path` a constant, which is not possible with the out-parameter approach.

ehmry, 2016-05-09:
Ok, new ``path_from_label`` at 89fcc9fdad811235bfcbf9968976d0903d55b140 with the signature
```
typedef Path<128> Label_path;
static inline Label_path path_from_label(char const *label)
```

Fs_log updated at 8b2e399bb97e4c2223df8ac3585dec10573549b2

nfeske, 2016-05-09:
I actually liked the fact that your function was a template, letting the caller decide for a max path length by defining a type at the caller side. How about letting the caller pass a `PATH` type as template argument? (very similar to your original version, but using a typename instead of a size_t as template argument)

```
template <typename PATH>
PATH path_from_label(char const *label)
{
  PATH path;
  ...
}
```

ehmry, 2016-05-09:
I have a template version, but my compiler can't deduce the typename without ``path_from_label<New_type>(label_str)``

nfeske, 2016-05-09:
Yes. The compiler cannot deduce the type automatically because it only looks at the signature, not the return value. But I think that passing the type manually is just fine. Don't you think so?

ehmry, 2016-05-09:
@nfeske Yea, I think thats fine, ``path_from_label`` isn't called very often, and always near the declaration of the return value.

chelmuth, 2016-06-22:
Merged the commits above to staging.

chelmuth, 2016-06-23:
@ehmry could you please have a look at the current staging branch and the following build error?
```
  Program server/fs_log/fs_log
    COMPILE  main.o
.../repos/os/src/server/fs_log/main.cc: In member function ‘virtual Fs_log::Session_component* Fs_log::Root_component::_create_session(const char*)’:
.../repos/os/src/server/fs_log/main.cc:95:17: error: use of deleted function ‘Genode::Path<128u>& Genode::Path<128u>::operator=(Genode::Path<128u>&&)’
        dir_path = path_from_label<Label_path>(tmp);
                 ^
In file included from /plain/krishna/src/genode/genode_staging.git/repos/os/src/server/fs_log/main.cc:15:0:
.../repos/os/include/os/path.h:338:15: note: ‘Genode::Path<128u>& Genode::Path<128u>::operator=(Genode::Path<128u>&&)’ is implicitly deleted because the default definition would be ill-formed:
 class Genode::Path : public Path_base {
               ^
.../repos/os/include/os/path.h:338:15: error: use of deleted function ‘Genode::Path_base& Genode::Path_base::operator=(const Genode::Path_base&)’
.../repos/os/include/os/path.h:27:15: note: ‘Genode::Path_base& Genode::Path_base::operator=(const Genode::Path_base&)’ is implicitly deleted because the default definition would be ill-formed:
 class Genode::Path_base
               ^
.../repos/os/include/os/path.h:27:15: error: non-static const member ‘char* const Genode::Path_base::_path’, can’t use default assignment operator
.../repos/os/include/os/path.h:27:15: error: non-static const member ‘const size_t Genode::Path_base::_path_max_len’, can’t use default assignment operator
.../repos/os/src/server/fs_log/main.cc:100:16: error: use of deleted function ‘Genode::Path<128u>& Genode::Path<128u>::operator=(Genode::Path<128u>&&)’
       dir_path = path_from_label<Label_path>(label_str);
                ^
.../repos/os/src/server/fs_log/main.cc:103:15: error: use of deleted function ‘Genode::Path<128u>& Genode::Path<128u>::operator=(Genode::Path<128u>&&)’
      dir_path = path_from_label<Label_path>(label_str);
               ^
.../repos/os/src/server/fs_log/main.cc:107:14: error: use of deleted function ‘Genode::Path<128u>& Genode::Path<128u>::operator=(Genode::Path<128u>&&)’
     dir_path = path_from_label<Label_path>(label_str);
```

chelmuth, 2016-06-23:
Maybe I made a mistake while cherry-picking but we also get
```
path_from_label/test-path_from_label
    COMPILE  main.o
.../repos/os/src/test/path_from_label/main.cc: In function 'int main()':
.../repos/os/src/test/path_from_label/main.cc:25:38: error: no matching function for call to 'path_from_label(Genode::Path<64u>&, const char*)'
  path_from_label(path, label.string());
                                      ^
.../repos/os/src/test/path_from_label/main.cc:25:38: note: candidate is:
In file included from /home/checker/checker.git/genode-package/repos/os/src/test/path_from_label/main.cc:15:0:
.../repos/os/include/os/path.h:379:21: note: template<class PATH> PATH Genode::path_from_label(const char*)
  static inline PATH path_from_label(char const *label)
                     ^
.../repos/os/include/os/path.h:379:21: note:   template argument deduction/substitution failed:
.../repos/os/src/test/path_from_label/main.cc:25:38: note:   cannot convert 'path' (type 'Genode::Path<64u>') to type 'const char*'
  path_from_label(path, label.string());
```

ehmry, 2016-06-23:
@chelmuth The problem is there was an operator for copy assignment (&) but no operator for move/temporary assignment (&&). Fixed at 6630f7123c16ffe6ff96d61fb838bd9a60192053.

chelmuth, 2016-06-23:
With 6630f71 *path_from_label* still produces the error above.

chelmuth, 2016-06-23:
Additionally `make run/libc_vfs_fs` produces
```
/repos/libports/src/lib/libc/file_operations.cc: In function ‘void resolve_symlinks(const char*, Libc::Absolute_path&)’:
/repos/libports/src/lib/libc/file_operations.cc:113:34: error: use of deleted function ‘Genode::Path<1024u>& Genode::Path<1024u>::operator=(const Genode::Path<1024u>&)’
   current_iteration_working_path = next_iteration_working_path;
                                  ^
In file included from /plain/krishna/src/genode/genode_staging.git/repos/libports/src/lib/libc/file_operations.cc:17:0:
/repos/os/include/os/path.h:338:15: note: ‘Genode::Path<1024u>& Genode::Path<1024u>::operator=(const Genode::Path<1024u>&)’ is implicitly deleted because the default definition would be ill-formed:
 class Genode::Path : public Path_base {
               ^
/repos/os/include/os/path.h:338:15: error: use of deleted function ‘Genode::Path_base& Genode::Path_base::operator=(const Genode::Path_base&)’
/repos/os/include/os/path.h:27:15: note: ‘Genode::Path_base& Genode::Path_base::operator=(const Genode::Path_base&)’ is implicitly deleted because the default definition would be ill-formed:
 class Genode::Path_base
               ^
/repos/os/include/os/path.h:27:15: error: non-static const member ‘char* const Genode::Path_base::_path’, can’t use default assignment operator
/repos/os/include/os/path.h:27:15: error: non-static const member ‘const size_t Genode::Path_base::_path_max_len’, can’t use default assignment operator
/repos/libports/src/lib/libc/file_operations.cc:186:16: error: use of deleted function ‘Genode::Path<1024u>& Genode::Path<1024u>::operator=(const Genode::Path<1024u>&)’
  resolved_path = next_iteration_working_path;
                ^
/repos/libports/src/lib/libc/file_operations.cc: In function ‘int fstatat(int, const char*, stat*, int)’:
/repos/libports/src/lib/libc/file_operations.cc:365:12: error: use of deleted function ‘Genode::Path<1024u>& Genode::Path<1024u>::operator=(const Genode::Path<1024u>&)’
   abs_path = cwd();
```
Any hints?

ehmry, 2016-06-23:
The problem now is the compiler isn't converting the operator arguments to const, this feature is looking more and more dubious.

Fixed at ebdc00b81cd5384c086a0f78c56179e6cf38573f

chelmuth, 2016-06-23:
The copy constructor for `T` should take a `T const &` parameter. We should investigate what goes wrong here...

chelmuth, 2016-06-23:
From a close look at the `Path<>` constructors, I'm sure some of them are missing a call to the correct base-class constructor `Path_base(char *buf, size_t buf_len,...)` because `Path_base::_path` and `Path_base::_path_max_len` are constant members.

chelmuth, 2016-06-23:
I removed the commits in this issue until we have compilable state. @ehmry as soon as you are ready with a new series I'll merge your work.

ehmry, 2016-07-01:
The move assignment troubles have gone away, rebasing against master...

commit 87401e98 (2016-06-22): server/fs_log: improve client isolation

Use a seperate handle at each session.
Use SEEK_TAIL to append messages to files.
Increase packet buffer.

Fixes #1777
commit 5f35f520 (2016-08-23): server/fs_log: improve client isolation

Use a seperate handle at each session.
Use SEEK_TAIL to append messages to files.
Increase packet buffer.
Refactor to component framework.

Fixes #1777
Issue #2060
closed 2016-08-30 by chelmuth