genodelabs/genode #1781: hw: review signal semantics in comparison the the generic implementation

chelmuth, 2015-11-18:
With the changes in #1738 esp. e32c19baa6292597cc7f996bc5b2fdce85d0bc5a, we revealed a gap between the generic signal-implementation semantics and the interpretation in base-hw. The generic implementation ensures correct reference counting in `Signal_context` according to the life time of `Signal` objects to support safe destruction of contexts. In base-hw, the `Signal` life time also triggers `Kernel::ack_signal()` if the last reference to the linked context is dropped. Unfortunately, the generic `Signal_context` object stores the current pending signal as a member of type `Signal::Data` which is just a plain data object not connected to the reference counting. In base-hw, we now have the challenge to demarshal the occuring signal from the UTCB in `Signal_receiver::block_for_signal()`, store it in `Signal_context::_curr_signal`, and only acknowledge at the kernel after is was retrieved from `_curr_signal` in `Signal_receiver::pending_signal()`.

From my investigation, the current issues with `run/nic_loopback` and `run/blk_cache` (at least) on staging originate from the described gap. @m-stein please add any information a missed from our offline discussion yesterday.

m-stein, 2015-11-24:
There are some more open issues in the signal framework like the redundancy of Signal_context::_pending and Signal_context::_curr_signal (may both be raplaced by Signal_context::_num) or that Signal_context::submit/Signal_receiver::local_submit shouldn't be public but I would vote for opening separate issues for them.

chelmuth, 2015-12-01:
@m-stein should we close this issue or remove the fixed label?

m-stein, 2015-12-02:
I forgot to add a "Fixed". Thanks.

closed 2015-12-02 by m-stein