genodelabs/genode #1782: VFS ram plugin loses files on move

ehmry, 2015-11-18:
Nodes get renamed without rebalancing the parent directory AVL tree.

ehmry, 2015-12-02:
@chelmuth Thank you for finding the lock error, I've managed to get rename to work just using guards. Is there any preperation to be done before removing locks from VFS plugins, or can they be phased out immediately?

chelmuth, 2015-12-02:
From what I learned just now, we had to adapt Noux to enter the VFS library only single-threaded. For multi-threaded native libc applications, I myself had to look into the code to be sure...

chelmuth, 2016-03-07:
@ehmry now we have two distinct commits fixing this issue. Could you please clarify how we should proceed with 0a8229f9f32bc80e915f49ae2195952fde81a82f and fa6f879b5b87a69ff369cc5ca88780ad854231a7?

ehmry, 2016-03-07:
Libc wants rename to replace files when the destination exists, so I updated the VFS ram plugin to do that, using my new rename from this issue as a base. If #1900 is merged then this issue can silently close, so I would say just worry about that one.

commit 23c1d4c6 (2016-03-07): libc, vfs: rename improvements

Perform POSIX rename checks at the libc VFS plugin.
Clarify rename at VFS fs, ram, single, and tar file systems.

Issue #1900
Fixes #1782
closed 2016-03-17 by chelmuth