genodelabs/genode #1784: File_system: support for file modification timestamps

cnuke, 2015-11-22:
By now having support for something like `st_ctim` and `st_mtim` sounds reasonable. Adding this feature would first of all involve extending `File_system::Status` as well as adding the required functionality to each file system or rather enabling the usage by tapping into the existing API (e.g. `rump_fs`) and probably the layers above (VFS). I am not sure if it is actually necessary to support it in all pseudo file system (or even `ram_fs` for that matter) but at least supporting in the ones used for persistent storage would be nice.

cnuke, 2019-03-18:
The plan for implementing the functionality is as follows:

1. Add a `Timestamp` member to `File_system::Status` which saves the modification time as signed 64bit integer. If a more structured type is needed, we will introduce convenience utilities.
2. Add a new packet type to the `File_system` session called `WRITE_TIMESTAMP` which allows clients to update the modification timestamp of a node. The rough plan is to send this packet whenenver a “dirty” node is closed or before a `SYNC` packets but that's up to the client, e.g. VFS/libc.

Setting the modification time of a node should be a deliberate decision of the client rather than an implicit side-effect of the file system provider, which may or may not has access to a time source.

Afterwards we'll deal with the VFS and libc side of things.

nfeske, 2019-03-18:
Thank you for capturing the outcome of our offline discussion, @cnuke.

cnuke, 2019-03-21:
My [file_system_modtime__wip](https://github.com/cnuke/genode/commits/file_system_modtime__wip) branch contains the first sketch (be warned I haven't even tried to compile it yet).

nfeske, 2019-03-25:
The commits already very nice. Please see my inline comments.

cnuke, 2019-04-10:
I updated my [wip](https://github.com/cnuke/genode/commits/file_system_modtime__wip) branch. It now contains a semi-functioning implementation, open issue to consider:

- For one it relies on doing the right thing at the client-side, i.e., calling `utimes()`. This does not happen a newly created files (that includes `cp`ing as well) as that is normally the job of the kernel .
- Apart from the that, in case of a libc component, the modification time always gets updated whenever `_vfs_sync()` is executed. Since we do not track if a file was opened r/o or rather if it was modified (the same is true for noux) we now change it accidently as a side-effect.
- We normally discourage writing to directories but we now might still update their modification time.
- Sophisticated VFS plugins, e.g., `vfs_rump`, might ignore our attempt to set the modification time and will change it own their own.

So far I will ignore the last two points and focus on the first two.

ehmry, 2019-04-11:
The eager sync has already been identified as a unnecessary and don't see any problems with skipping it for unmodified files.

cnuke, 2019-04-11:
Yes and the tracking of unmodified files is the thing missing so far.

cnuke, 2019-04-18:
I've force pushed the branch again and it now contains commits that address the first two points of my previous post. Together with some small adjustments in the *genodians.org` appliance, lighttpd serves resources with the proper modification time now. I'll work on cleaning it up for staging next week.

cnuke, 2019-04-23:
The current WIP state is now live on *genodians.org*.

cnuke, 2019-04-24:
Running it on *genodians.org* revealed a bug in the current implementation (I didn't catch the Insufficient_buffer exception of the `Fs_file_system` in the libc) and I changed it to use the same mechanism as the `sync` packet now. That being said, I reverted *genodians.org* back to an older version b/c there are still some issue left.

cnuke, 2019-05-07:
I put the commits on top of the current master branch and they are live again on *genodians.org*.

@m-stein you may give thunderbird's RSS tool a try.

m-stein, 2019-05-07:
@cnuke Ah thanks!

m-stein, 2019-05-07:
@cnuke Seems to work - thanks a lot!

cnuke, 2019-05-16:
And the mtime commits are off again, there are apparently still some problems left (*genodians.org* sometimes still gets stuck, setting verbose="yes" in noux seems to make it more reliable while interactive usage works fine).

nfeske, 2019-09-23:
The patch series looks excellent. I just rebased it on top my current libc branch (https://github.com/nfeske/genode/commits/libc) ... to be merged to staging soon.

chelmuth, 2019-09-26:
The message in commit e3f29396fb044e80835be5bea83f6078508134b8 states

> Up to now Noux used a monotonic clock whose initial start value always was '0' (which correlates to the start of the UNIX epoch) to provide a notion of time. In addition it is now possible to use the RTC session to set the initial value for use-case where having a proper real world time matters. *It does not do any time-zone conversion - the returned time depends on the component provide the RTC service*.

This seems to hint that the RTC service should do time-zone conversion, which is not the case in our current implementation where the system time is always UTC. The time-zone handling is done in the clients. Please correct me if I'm wrong or suggest a comment update.

cnuke, 2019-09-26:
The sentence should just make clear that it takes the *RTC* value as-is. Thanks for pointing out loose phrasing, I don't mind removing it entirely.

cnuke, 2019-09-27:
Sorry, I forgot to annotate the commit yesterday but that reminded me to document the value member as well. Commits 2131375 and 823bee2 should address @chelmuth remarks.

nfeske, 2019-09-30:
@cnuke Thanks for the fixups. I merged them (slightly redacted - maybe you can give them a look again) to staging.

nfeske, 2019-11-19:
Fixed in master.

closed 2019-11-19 by nfeske