genodelabs/genode #1787: Deprecate ROM filename

ehmry, 2015-11-23:
As mentioned in #1766, the filename argument can be removed and the rightmost session label element treated as the name of the ROM. By the same logic the root argument on File_system sessions could go the same way, such as ``init -> child -> /``.

nfeske, 2015-11-23:
I am not sure if the same reasoning applies to the root argument of file-system sessions. The root argument allows the client to use knowledge about the file system hierarchy for limiting its own view of the file system. As far as I can see, in contrast to the label, the root argument is not meant as a criterion for taking session-routing decisions or for selecting a policy at the file-system server.

ehmry, 2016-01-12:
I've started running into problems with ROM session request buffer overruns from large names being duplicated between the label and filename arguments, so I went ahead and removed the filename argument from ROM session. I wouldn't say that this is a priority issue for me, but I think it does make things a bit simplier.

The following replaces the previous two commits.

ehmry, 2016-05-12:
rebase

chelmuth, 2016-05-13:
You're right, we should consider to integrate the cleanup in the next release.

chelmuth, 2016-07-01:
I'd like to merge this cleanup.  @ehmry  could you please squash and rebase these changes on current master?

ehmry, 2016-07-05:
Rebased against master. fba1454e3fdb95b1c6aa577ee657bc2d03ce00c7 is dangerous because it changes the result of ''Session_label(args)'', but I think it is worth it to decouple Session_label from Session_args.

I'm adjusting my Nix work to run against master, and I'll need this to clear to get it to work, otherwise I have to worry about rewriting both label and filename on ROM session requests.

ehmry, 2016-07-05:
@nfeske I've split the compound label constructor to a prefixed_label function as you suggested.

nfeske, 2016-07-06:
I added the first commit ("Move Session_label from os to base") to staging. @ehmry please review my fixups.

nfeske, 2016-07-06:
I also added the second part of the change to staging, with the only reservation being the part I annotated inline. @ehmry I would very much appreciate a fixup regarding this part. I nevertheless put it to staging to see what the buildbot/autopilot is saying.

ehmry, 2016-07-06:
I've cleanup the ROM label rewriting at fca6e11d8af20fe1154cdd0adeaf28b20b31eab9

ehmry, 2016-07-06:
fca6e11 needs fixup 8b8d7a1 :(

nfeske, 2016-07-06:
Thank you for reworking that part. I added both fixups to staging.

nfeske, 2016-07-07:
I added a couple of minor fixups to the staging branch. As I was still not fully satisfied with the `Child_policy_redirect_rom_file`, I tried to simplify it by adding a `Session_label::prefix` method (commit 78dae090311598d5dbb9f4de62085aac254759b6). @ehmry Do you approve?

ehmry, 2016-07-07:
Yes, I can think of other situations where ''::prefix'' can be useful.

nfeske, 2016-07-08:
The never-ending story goes on. ;-) I added a few more fixups to staging. Apart from the usual fixes of the automated tests, I changed the new API functions to avoid the use of pointers. This way, we can successively replace the occurrences of `char const *` by `Session_label const &`, e.g., in the connection constructors. It also allows us to use `==` for comparing strings instead of calling the ugly `strcmp` function. In the end, we hopefully end up in a situation where the `String::string()` method is only very rarely needed. The specific commit is 3ff5019bb0baaaa60043973437c939caf816eb9d

closed 2016-07-12 by nfeske

nfeske, 2016-07-12:
@ehmry Now that this change entered master, it might be a good time to adapt the components of genode-world to it.

ehmry, 2016-07-21:
I have a small nitpick for the prefix function, d499b291b9d87f11fd9e12252425ea61999f549b

ehmry, 2016-07-21:
Swapped the booleans around.

commit 7982bb58 (2016-07-06): remove 'filename' from ROM sesion args

Conveying the ROM filename as the final label element simplifies
routing policy and session construction.

Fixes #1787
commit 2b8c1af9 (2016-07-11): remove 'filename' from ROM sesion args

Conveying the ROM filename as the final label element simplifies
routing policy and session construction.

Annotations by nfeske:

This commit also changes the ROM session to use base/log.h instead of
base/printf.h, which produced build error of VirtualBox because the
vbox headers have a '#define Log', which collides with the content of
base/log.h. Hence, this commit has to take precautions to resolve this
conflict.

The commit alse refines the previous session-label change by adding a
new 'Session_label::prefix' method and removing the use of 'char cons [... cut]