genodelabs/genode #1788: Unnoticed updates between session creation and signal-handler registration

chelmuth, 2015-11-23:
For example, in *report_rom* a server-side ROM update between session creation and signal-handler registration may go unnoticed. This is especially bad if the client implementation depends on the update signal to occur even for the initial update. The proposed solution is to notify the client initially in the signal-handler registration component function to enforce a client-side ROM update.

This issue affects other session interfaces supporting signal handlers, e.g, terminal session.

nfeske, 2017-02-28:
Triggering a signal unconditionally at the signal-handler registration has a potentially bad effect: If the client responds to the signal by re-creating the entire ROM connection (which may be conveniently wrapped in a `Reconstructible`) and re-registering its signal handler, an infinite feedback loop is created.

Hence, I propose to address this problem by server implementations individually. As a general pattern to deal with this problem:

* The server tracks the ROM versions via a simple counter variable that is increased each time the ROM gets updated.
* At session creation time and when the update RPC function is called, the server remembers the version of the handed-out ROM in the session-component object.
* When the sigh RPC function is called, the server compares the handed-out version with the current version. If they differ (and only if), it triggers a signal to encourage its client to request the now-current version.

chelmuth, 2021-06-23:
The suggested solution is not completely implemented in *repos/os/include/report_rom/rom_service.h* and, thus, *report_rom* unconditionally notifies clients on signal-handler registration.

closed 2022-02-11 by chelmuth