genodelabs/genode #1791: vbox: 64-bit guest support

cproc, 2015-11-24:


cproc, 2015-11-24:
@alex-ab: my updated seoul branch includes the AHCI controller model commit from your genode_15_08 branch, which was not referenced in the seoul.port file. Is this ok?

alex-ab, 2015-11-25:
@cproc: great work !
I adjusted the links as suggested by @chelmuth.

The vbox_auto_win81_64.run script seems not to succeed for me on the test machine, since in the log the sequence ´´´Guest Additions capability report: seamless: yes, hostWindowMapping: no, graphics: yes``` does not appear. (Still Win81/64bit is up).

@cproc: did it work for you, and if yes, do you have an idea ?

cproc, 2015-11-25:
@alex-ab: the last time I executed the run script on the x250 yesterday, it succeeded. But I'm not sure if I had the VDI auto-reset feature enabled at the time or not, so there might have been differences with the content of the VDI images (guest additions?).

chelmuth, 2015-11-25:
I merged commit and fixup to staging but are still wondering why the commit touches seoul. Does there exist any interdependency?

cproc, 2015-11-25:
Seoul has its own definition of the UTCB structure, which must be kept in sync with the one in NOVA.

chelmuth, 2015-11-25:
Okay, nevertheless I suggest to split the commit into two: one for the NOVA extension (incl. seoul adoption) and one for the vbox extension tomorrow.

cproc, 2015-11-26:
@chelmuth: I've split the commit as suggested. The two new commits replace the original one. I also updated the VDI image on the autopilot test machine. The log message which triggers the success of the run script only appeared after user login, so I removed the password of the user and now it logs in automatically.

chelmuth, 2015-11-26:
@cproc I'll merge the work in a second...

commit 2ff4f50a (2015-11-25): vbox: 64-bit guest support

Fixes #1791
commit 1a9c9629 (2015-11-26): vbox: 64-bit guest support

Fixes #1791
closed 2015-11-30 by cproc