genodelabs/genode #1797: hw: kernel assertion with run/rump_ext2 on hw_wand_quad

chelmuth, 2015-11-26:
The test breaks nightly with
```
[init -> ram_blk] Using file=ext2.raw as device with block size 512.
Assertion failed: !Descriptor::valid(_entries[i]) || _entries[i] == e
  File: .../repos/base-hw/src/core/include/spec/arm/short_translation_table.h:233
  Function: void Genode::Page_table::insert_translation(Genode::addr_t, Genode::addr_t, Genode::size_t, const Genode::Page_flags&)
```

chelmuth, 2015-11-27:
The issue can also be triggered with `make run/bomb` on this platform.

skalk, 2015-12-02:
the problem here is that on ARM we missed to remove the memory-mapped I/O regions, which are mapped one-by-one for the kernel, from core/kernels virtual memory allocator - in contrast to the x86 tree, where this was done correctly by our Muen friends ,-)

skalk, 2015-12-03:
For reasons: the rump_ext2 test is still failing on this platform because it is damn slow. One "round" in this test lasts about 2 seconds on hw_imx53, but 30 seconds on hw_wand_quad. I cannot imagine why. Maybe memory accesses are that slow?

skalk, 2015-12-07:
Ok, last friday I realized that the wandboard's L2 cache is disabled, which explains why it is that slow. I've enabled it with the consequence that the first new data access after the MMU got enabled leads to a stall system. I've opened a new issue #1807 for this.

commit 1b418a47 (2015-12-02): hw: remove 1:1 I/O mappings from virtual memory

Likewise on the x86 branch, we have to remove all virtual memory ranges from the
virtual memory allocator that are used by one-by-one mappings of I/O regions
used by the kernel.

Fix #1797
commit ee461968 (2015-12-10): hw: remove 1:1 I/O mappings from virtual memory

Likewise on the x86 branch, we have to remove all virtual memory ranges from the
virtual memory allocator that are used by one-by-one mappings of I/O regions
used by the kernel.

Fix #1797
closed 2015-12-10 by skalk