genodelabs/genode #1801: report_rom: partial ROM request matching

ehmry, 2015-11-30:
It would be nice to have, especially when you run into things like ``bash.2 -> ... -> wlan_access_points``.

chelmuth, 2015-11-30:
The suffix match is pretty weak - almost too weak from what I remember while discussing your initial label-matching patch with @nfeske. I'd feel much better if you would use a combination of a label prefix for the (maybe complex) subsystem the policy matches and a label suffix for the resource a.k.a. ROM. Would that be feasible?

ehmry, 2015-11-30:
Yes, that works with this patch, prefix and suffix.

nfeske, 2015-11-30:
That's a good use of the new session-label utilities!

But while we are at the report-rom configuration, I am wondering if it was a mistake to introduce the intermediate `<rom>` node in the first place? It is always there and there is no other sibling node anyway. What do you think? Would it be sensible to allow the specification of policies directly at the config nodes like every other modern component is doing it? We could still probe for a `<rom>` node for a while (like one release cycle) until dropping it later.

ehmry, 2015-11-30:
I don't see why not, I think policy is a good term and a good pattern.

chelmuth, 2015-12-01:
Sounds like a feasible plan. @ehmry could you please adjust your patch regarding the discussion? I also have a tiny remark: The `Session_policy` constructor supports to pass the config node as an argument, which should also work for the `<rom>` node. If I'm not mistaken this spares us to copy the best-match lookup into report_rom.

ehmry, 2015-12-04:
New patch to apply policies from the top-level and fallback to policies at ``<rom>``.

chelmuth, 2015-12-10:
@ehmry I merged your line of work. Please review my changes in d72c4df86dccf7b8eb383a2589e5ce65b13e26c2.

ehmry, 2015-12-10:
@chelmuth :+1:

ehmry, 2015-12-10:
The debugging code is still in the server BTW.

https://github.com/genodelabs/genode/blob/d72c4df86dccf7b8eb383a2589e5ce65b13e26c2/repos/os/src/server/report_rom/main.cc#L41

chelmuth, 2015-12-10:
@ehmry your messages reflects that my commit annotation is misleading. What I actually removed was the `verbose="yes"` config attribute in the run script.

commit 4ec1c684 (2015-12-10): server/report_rom: partial ROM label matching

Fixes #1801
commit 6e9fa10e (2015-12-17): server/report_rom: partial ROM label matching

Fixes #1801
closed 2015-12-17 by chelmuth