genodelabs/genode #1802: intel_fb_drv: build only on suitable platforms

cnuke, 2015-11-30:


chelmuth, 2015-11-30:
We left out the `pci` spec by intention as the driver should build also on base-linux as is the case for `wifi_drv`. You're right that it makes no sense to run it on Linux currently but this is the responsibility of the integrator. @nfeske do you have any additions?

nfeske, 2015-12-01:
That's right. Building the driver for Linux does no harm but may eventually reveal problems related to the build (e.g., conflicting defines). Actually, I think that there is no fundamental contradiction. It's just that nobody has stepped up to facilitate the use of Linux' user-level device-driver API under Genode, yet. ;-)

chelmuth, 2015-12-01:
Okay, I close the issue then.

closed 2015-12-01 by chelmuth