genodelabs/genode #1804: hw: remove main thread's initial UTCB from virtual memory

skalk, 2015-12-02:
Currently, the main thread's initial UTCB is not cleared from the corresponding RM session not even in core, which leads to "double insertion" exceptions when trying to map something to this area (e.g. pf in core when running bomb test on hw_x86_64). We should simply take out the initial UTCB area from the ranges used by the RM sessions.

m-stein, 2015-12-03:
I would vote for a short explanation for why we have choosen this address in the commit message as well as in the source code. A hint on why we need to choose the address statically anyway would also be useful, I think (this should have been documented already by me when implementing the function in the first place). Apart from that I'me fine with the commit.

skalk, 2015-12-07:
@chelmuth I followed m-stein's suggestion, please add the fixup commit 3b12e52a1fef1bdbd901189c189b78a4c2547295 to staging. It includes a commit message addon within its commit message.

chelmuth, 2015-12-07:
Okay, both commits are now squashed and on top of staging to preserve the commit message update. Please check.

skalk, 2015-12-07:
@chelmuth fine, thanks!

commit 5838de73 (2015-12-02): hw: remove main thread's initial UTCB from vm area

Fix #1804
commit 6fa70eec (2015-12-04): hw: remove main thread's initial UTCB from vm area

Fix #1804
commit b8b57c0a (2015-12-07): hw: remove main thread's initial UTCB from vm area

The main thread's UTCB, used during bootstrap of the main thread before
it allocates its context area, needs to be outside the virtual memory
area controlled by the RM session, because it is needed before the main
thread can access its RM session.

Fix #1804
commit 2316937e (2015-12-10): hw: remove main thread's initial UTCB from vm area

The main thread's UTCB, used during bootstrap of the main thread before
it allocates its context area, needs to be outside the virtual memory
area controlled by the RM session, because it is needed before the main
thread can access its RM session.

Fix #1804
closed 2015-12-10 by skalk