genodelabs/genode #1806: screencasts of newly released features

dckc, 2015-12-03:
See https://github.com/genodelabs/genode/issues/1000#issuecomment-161410556

m-stein, 2016-02-10:
I've ported avconv and the avdevice lib and currently trying to feed the fbdev input of avconv with an interposed framebuffer session.

dckc, 2016-02-10:
Interesting... it didn't occur to me that you'd have to port screen-capture tools to genode to do this.

m-stein, 2016-02-11:
Of course, it's only one option to do that. As parts of the avlib are already ported (avplay, avcodec, avfilter, avformat, avutil), I think it's a low-hanging fruit and has the benefit of bringing a broad feature set. Apart from the screencast topic, it can be used to convert all kinds of video and audio streams. I also thought about re-implementing how avconv utilizes libav to get a screen-capture component from scratch. But I've decided aginst it for now.

m-stein, 2016-02-11:
@dckc Maybe you can share your idea?

dckc, 2016-02-11:
I didn't really have an idea. I hadn't thought it through. But as I mull over it now...

When I ran the genode Aurora example, I didn't boot into genode. Genode ran under qemu or something in linux and it created one window; it didn't take over my whole screen. Likewise, the build process was just one of many windows on my machine.

I would expect that normal screen cast recording tools running outside genode could capture what's going on in the genode build and run windows.

But I have no direct experience with these tools.

m-stein, 2016-02-23:
@chelmuth I've uploaded a state [(branch)](https://github.com/m-stein/genode/tree/1806_screencasts_of_newly_released_features) that might be added to the upcoming release as port of avconv. This state can be used also to create screencasts of the Turmvilla scenario. I'll provide details about that in my corresponding release notes contribution.

dckc, 2016-04-07:
I see 16.02 is out. Did anybody happen to record the process of grabbing the code and getting it going?

dckc, 2016-04-07:
I made a 6 minute screencast that shows getting the toolchain and following the getting started docs up to `make run/demo`.

 - [genode-16.02-demo.ogv](https://drive.google.com/a/madmode.com/file/d/0BxOfJ7sv3AmqUXVzVlFucVlhT1k/view?usp=sharing) 77MB

It could probably be edited down to something shorter, but it's something.

ehmry, 2016-04-08:
Can you upload it somewhere public?

dckc, 2016-04-08:
I thought I did... oops! I screwed up the sharing settings. Try again?

ssumpf, 2016-04-08:
@dckc: It works now ;)

dckc, 2016-09-15:
the .iso and .ova files for 16.08 are great... I got the seL4 example scenario running in just a few minutes. The hardest part was figuring out that all I needed to do was
 - get the .iso flie
 - get the .ova file; open it to start virtualbox
 - use settings to point the ide controller to the .iso file
 - start the VM
and voila!

chelmuth, 2019-03-14:
Screenshots can now be done via http://genodians.org/ehmry/2019-01-31-flif_capture

m-stein, 2019-03-14:
The initial ``avconv`` already worked and I also used it to publish a pretty fluent [Genode screencast](https://www.youtube.com/watch?v=QeBvog_ZoXM). The only thing that kept us from merging the port was that it uses two _ioctls_ that were implemented the old way and we planned to introduce a new way to do this (via a common file for each ioctl). Has the new way, in the mean time, become stable and should I try to rebase the port?

cnuke, 2019-03-14:
No, there is, however, a proof-of-concept implementation on my [libc_ioctl_fs](https://github.com/cnuke/genode/commits/libc_ioctl_fs-2018-11-05) branch.

m-stein, 2019-03-14:
@chnuke A pity... but thanks for the update! Do you see the proof-of-concept make its way to master in the near future? I think, especially in the context of our current priorities, screencasts would be a really helpful thing to have.

cnuke, 2019-03-14:
I do not plan to actively pursue it for the time being (the commits might need to be adapted for the recent VFS changes but should work so far - the Device_file_system needs to be reworked to be fit for upstreaming it though).

Regarding creating screencasts in Sculpt, you could easily provide a pkg archive whose bin archive contains all the changes you might need in the binaries as interims solution.

m-stein, 2019-03-26:
I've rebased my old avconv work to staging and it the screencast demo is almost running again (using my lxfb file system addition for vfs/libc):
https://github.com/m-stein/genode/commits/1806_screencasts_of_newly_released_features

Next, I'll try to create source archives and a package out of this and then to deploy it on my sculpt machine.

m-stein, 2019-03-27:
The demo runs now and avconv seems to read the framebuffer regularly with the right screen settings but for some reason it doesn't produce a valid output file nor does it finish after the configured time:
https://github.com/m-stein/genode/commits/1806_screencasts_of_newly_released_features
https://github.com/m-stein/genode-world/commits/1806_screencasts_of_newly_released_features

nfeske, 2021-02-26:
We side-stepped the technicalities by using Sculpt with an HDMI grabber.

closed 2021-02-26 by nfeske