genodelabs/genode #1808: hw: optimize capability destruction

skalk, 2015-12-07:
When working on issue #1623, nfeske came up with an alternative approach to the current solution. Instead of acknowledging received capabilties to the kernel, the kernel could run through all UTCBs od a PD when a capability should be destroyed. Every UTCB that is dirty, meaning the kernel has written capabilities to it, which the user-thread did not processed right now - flagged by a bit, is inspected whether it contains the capability that should be destroyed. Only if no UTCB contains the capability it really gets destroyed.
This solution increases the costs for destroying capabilities (iterating through UTCBS), but decreases the costs of receiving a capability (syscall). It is assumed that receiving capabilities occurs more often, and that the syscall costs are higher than the UTCB iteration and inspection. Of course this has to be proved when implementing this test-wise.

nfeske, 2015-12-07:
Thanks for the conserving the idea in this form!

nfeske, 2021-02-26:
I'm closing this issue because it is ripe to be archived.

closed 2021-02-26 by nfeske