genodelabs/genode #1812: libports: add libpq and libpqxx libraries

alexygallar2, 2015-12-07:
Libraries for connection with database PostgreSQL server

chelmuth, 2015-12-07:
Are you working on this topic? IMO it would be best if you provide your port via a Genode repository on your side. The functionality is quite orthogonal to the Genode base system. For an example have a look at https://github.com/nfeske/genode-world or https://github.com/ehmry/genode-emery.

nfeske, 2015-12-07:
As the number of higher-level ports grows, we will need a way to keep track of what is available. I see two principle directions:

1. We might consider moving genode-world to the genodelabs account and integrate such ports there. Since it is separate from the base system repository, we would not need to be as strict with accepting and reviewing contributions as we are for genodelabs/genode.

2. We may provide an index of available repositories at our website, e.g., under the "community" section. This option seems to be is less convenient for the users of Genode though as they will eventually need to "hunt" for dependencies. It also increases the risk for duplicated efforts, compared to a situation where all developers use the same repository as reference.

What are your opinions? @ehmry May you share your's?

Personally, I have a slight preference towards the first approach since it has a better chance to create a sense of community and it fosters the review of commits by the joint user/contributor base.

alexygallar2, 2015-12-07:
>
> Are you working on this topic? IMO it would be best if you provide 
> your port via a Genode
Yes, I'm working on this issue. I already made the port and the libpq 
libpqxx libraries.
>
> repository on your side. The functionality is quite orthogonal to the 
> Genode base system. For an example have a look at 
> https://github.com/nfeske/genode-world or 
> https://github.com/ehmry/genode-emery.
Your recommendation is that the contributions organize them into a new 
repository?

alexygallar2, 2015-12-07:
The idea is to perform database query to external Postgres servers and not to make the Postgres port completely for now.

alexygallar2, 2015-12-07:
> As the number of higher-level ports grows, we will need a way to keep
> track of what is available. I see two principle directions:
>
> 1.
>
>     We might consider moving genode-world to the genodelabs [...]
I also believe that the first option is more convenient. So all 
contributions would have a single repository.

ehmry, 2015-12-07:
I would use a community repo, but I would still keep a personal repo. I think development is much easier when you can commit and changes branches idependently of the main repo.

chelmuth, 2015-12-08:
The idea of a *genode-world* repo under @genodelabs sounds attractive in my ears. As Norman already hinted I also see the higher chance of community-driven improvements and maintenance. In the end, it's far more easy to update the base and world repository from upstream on a new release and get compatible states than with scattered repositories. But is this really what developers can expect from genode-world? Are there more expectations we like to meet or exclude? In my opinion we need a clear statement what the repo is all about and how the *merge process* is handled. We may also discuss what mechanics can be added to the repo to find (or check for) a known-good revision of the base repository, e.g., the latest release, for the current genode-world revision.

nfeske, 2015-12-16:
I have moved my genode-world repository over to genodelabs/world.git now: https://github.com/genodelabs/genode-world

@alexygallar2 Please consider contributing your new libraries to this new repository.

alexygallar2, 2015-12-16:
Ok, then I'll try to add my contribution to this repository.
@nfeske : Should we follow the same procedure we use with Genode to add our contributions?

nfeske, 2015-12-16:
@alexygallar2 This would be perfect: Open an issue at the genode-world repository and refer to a working branch or individual commits to be found in your fork.

chelmuth, 2015-12-18:
Moved to genodelabs/genode-world#1

closed 2015-12-18 by chelmuth