genodelabs/genode #1819: Handling of bool attribute values in XML generator

chelmuth, 2015-12-11:
Currently, `Xml_generator::attribute("foo", true)` produces `foo="1"` which is parsed as `false` by `Xml_node::attribute_value("foo", false)`. The commit below reproduces and fixes the issue but drafts additional code to handle the ambiguity of `bool`, `long`, and `int`. I'm not sure if adding more and more overloads is the right direction. (This is already the case with `ascii_to`). The issue was detected in intel_fb_drv which uses `attribute` for `width` etc. as well as `connected`.

nfeske, 2015-12-16:
In principle, the functionality is the reverse of the ascii_to overloads. So maybe, we could think about introducing a set of from_ascii functions that match the encoding of the ascii_to counterparts. Then we could turn the `attribute` into a template that would just call the matching overload of `from_ascii`. That said, we could first go for the extension of the Xml_generator to solve the immediate problem. Later, when we replace those `attribute` overloads with a single template, we won't break API compatibility.

nfeske, 2015-12-16:
Oops. I mixed up the name of the to_ascii function. Please note the corrected comment above.

chelmuth, 2015-12-16:
@nfeske I implemented the overloads in `Xml_generator` and extended the test in 2616933a6bffb15e316f865d76eea747144f3556.

nfeske, 2015-12-16:
Very good! It will be just perfect if you move the adaptation of the decorator_stress test to a separate commit. For considering the joint use of the conversion functions from Xml_generator and the printf implementation, it would be nice to open a dedicated issue.

commit ca36f328 (2015-12-16): xml_generator: support bool and integer types

The Xml_generator is extended by overloads for bool as well as signed
and unsigned int, long, and long long.

Fixes #1819
commit 14c83805 (2015-12-17): xml_generator: support bool and integer types

The Xml_generator is extended by overloads for bool as well as signed
and unsigned int, long, and long long.

Fixes #1819
closed 2015-12-17 by chelmuth