genodelabs/genode #1833: editing config doesn't update anything (e.g <configfile>)

NobodyIII, 2015-12-24:
I built the Turmvilla scenario, and I edited the config files using vim (in noux), but nothing happened. I tried changing the window colors in multiple files, I tried adding the background grid (I disabled it in the default config), and I tried editing the wifi_drv config. In all of these situations, the updated files saved successfully and persisted when I reopened them, but there was no visible effect. I checked the logs as I tried these config changes, but nothing showed up to indicate that the programs received the configuration updates.
In case it helps, I am using the master branch (updated today), with a few cherry-picked commits for the Turmvilla scenario and for VGA console debugging.

nfeske, 2015-12-24:
Can you please double-check that you have cherry-picked the following commit? https://github.com/nfeske/genode/commit/1e4aacc51a31654438ef0fa9a3fde6e552b87811

NobodyIII, 2015-12-24:
Thanks! That fixed it.

closed 2015-12-24 by NobodyIII